mantidproject/mantid #14524: Kernel::Logger::is() not working as expected

FedeMPouzols, 2015-11-19:
I'm not sure if this is a confusion but several algorithms send debug messages to the logger only if...
```
if (g_log.is(Kernel::Logger::Priority::PRIO_DEBUG)) ...
```
That doesn't seem to work as expected. Try to open Mantid, set for example `information` level, and check what you get from these ones for example:
```
auto prio1 = g_log.is(Kernel::Logger::Priority::PRIO_DEBUG);
auto prio2 = g_log.is(Kernel::Logger::Priority::PRIO_ERROR);
auto prio3 = g_log.is(Kernel::Logger::Priority::PRIO_INFORMATION);
auto prio4 = g_log.is(Kernel::Logger::Priority::PRIO_CRITICAL);
auto prio5 = g_log.is(Kernel::Logger::Priority::PRIO_FATAL);
```

All the `prio` values are `true` (checked on win7 and debian).

The documentation of `Logger::is` says `Returns true if at least the given log level is set`. Does that "set" mean "selected by the user" or "available"? It seems to behave as the second but the code elsewhere seems to be using as if it should behave as the first option.

Related to #14467, where the behavior of the logger here was really awkward: https://github.com/mantidproject/mantid/blob/master/Framework/CurveFitting/src/CostFunctions/CostFuncLeastSquares.cpp#L410

martyngigg, 2015-11-20:
It's supposed to behave like the as your first case so that you can avoid constructing an expensive log statements if it is not going to be used.

I've marked this high priority for this release as this is a regression that we've introduced somewhere and will impact on performance in some areas.

NickDraper, 2015-11-20:
This might be interesting, when you set the level in mantid plot you are setting the level for that output channel.  The is will not know about the levels for channels, it will get the level of the root logger.  This is set in mantid to be the lowest of all of the channels (as it has to be for the channels to work).

So it might not always follow that this would follow a particular channel, but should follow the lowest of all of the channels.

NickDraper, 2015-11-20:
It is possible that the tracking of the root logger level to the minimum level of the channels is not working perfectly

PR #16123 (2016-05-13): Allow log levels to match the lowest of all filter channels
The log level of all the loggers is now set to follow the lowest log level of all of the FilterChannels (File, Console and message display in mantidplot).  It used to track them going down to debug, but never back up again, now it does.

**To test:**
This is tested pretty well in the unit tests in ConfigServiceTest.  However you should also start up Mantidplot, change the log level of the log window and check the correct level filter is applied when running algorithms.

Fixes #14524 

##### RELEASE NOTES
Does not need to be in the rel [... cut]
closed 2016-05-13 by peterfpeterson