mantidproject/mantid #14557: Importing a TBL file created from the v1 Refl GUI to the v2 gui should be easier

OwenArnold, 2015-11-24:
Version 1 of the GUI (Max's) works of a TBL file. These can be saved and loaded and effectively remember the entries in the table.

In Version 2 of the GUI, we work off TableWorkspaces. However we also introduced the ability to import a TBL file from the menu. However the import does not fully complete the import. You have to do an import as one step and a load as another. We should just be able to Import the table in one-shot.
[image]

OwenArnold, 2015-11-24:
@raquelalvarezbanos This sounds like the kind of thing that Matt could do. So reassign to him if you like.

raquelalvarezbanos, 2015-11-25:
@MattKing06 I reassigned this to you.

PR #14709 (2015-12-11): Importing ReflTBL will now load it into processing table in one go
Previously users had to import the TBL file, which loaded it into the ADS, and then open the table so that it was present in the processing table. Now importing a TBL file will load the table into the ADS and the processing table in one go.

**Code Review**:
- This is currently done based on the string containing the algorithm name in `showAlgorithmDialog`
- if the the algorithm is `LoadReflTBL` then we bring in the loading functionality
- this `showAlgorithmDialog` function is used elsewhere with different algorithms in the Ma [... cut]
closed 2015-12-11 by OwenArnold