mantidproject/mantid #14561: Document what Mantid reads from Nexus files

Anders-Markvardsen, 2015-11-24:
Add documentation which explains what information Mantid is reading as part of loading Nexus files. Perhaps most importantly the ISIS and SNS Nexus files.

This includes documenting LoadNexus and how it chooses which load algorithms to use. 

Add this documentation in the rst of the loaders.

KarlPalmen, 2015-11-24:
Dear Anders

I had a quick look at the code that decides which Nexus loader to use and there appears to be two different ways this is done

(1)    A way used by Load, which uses the NexusDescriptor and calls the confidence methods of the loaders

(2)    A way used by LoadNexus, which does not have a confidence method and therefore, I do not think is ever called by Load.

Karl

KarlPalmen, 2015-11-25:
This ticket is blocked by : 
 - [ ] #14572 
 - [ ] #14573

KarlPalmen, 2015-12-04:
In the NexusFile document of concepts, there is a section about structure that just has text "to be completed". This section would be very useful for #14572 & #14573 as well as this issue.
So I'll create an issue for this.

KarlPalmen, 2016-03-08:
Work is currently being done via issue #14940.

KarlPalmen, 2016-05-11:
Now awaiting #16070 to be either closed or moved to release 3.8 .

PR #16153 (2016-05-13): Nexus Loader Documentation Improvements
Improve Nexus Loading Documentation, showing how various types of data are copied from Nexus file to Workspace or other Output properties.

**To test:**

Check the release notes and that all work for this issue has been completed.

Fixes #14561 .

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the coding standards? Is it well structured with small focussed classes/meth [... cut]
closed 2016-05-13 by FedeMPouzols