mantidproject/mantid #14571: Cannot use Python Debugger in PyExec but in PyInit

picatess, 2015-11-25:
Using PTVS on Visual Studio it is possible to step into the PyInit method of python algorithms but not into PyExec.

__To Reproduce__

1. Select a system test which calls a python algorihtm
2. Set breakpoints in PyInit and PyExec
3. Run the system test
4. Attach the debugger to the python process ( You might have to use time.sleep(??) to attach). Note that for the python debugger in VS2015 to work you need to have PTVS. Make sure that Python debugging is enabled (Debug>Attach To Process > Attach to: Python code) and that the Python debug symbols are loaded (in  Tools > Options> Debugging> Symbols set teh .pdb location to the python2.7 direcotyry of teh ThridParty libs.)
5. You should see that the debugger stops in PyInit, but not in PyExec

martyngigg, 2020-02-25:
@DavidFair I assume this is probably a similar issue to one you recently closed about python debugging?

DavidFair, 2020-02-25:
Yeah, thanks for pointing this one out

Closing as this is a tooling limitation, the debugger can't set breakpoints across threads and unfortunately we can't really fix this one without using different tooling (or waiting for it to update)

closed 2020-02-25 by DavidFair