mantidproject/mantid #14595: Usage example for LoadTOFRawNexus

KarlPalmen, 2015-11-26:
The documentation needs a usage example.

Related issue #14577 .

NickDraper, 2016-02-19:
We need an example for this, are there any suitable files at the SNS that are small enough to go into the usage data?

Assign on if this is to the wrong person.

stuartcampbell, 2016-02-23:
@peterfpeterson Is this algorithm even used anymore ?

KarlPalmen, 2016-02-24:
Yes I'd like to know this. If the answer is no, it would save me having to document the behaviour of this algorithm in more detail. I yes, I'd like the usage example to resemble this usage (but be smaller). It may be used for my documentation.

KarlPalmen, 2016-06-29:
I found 3 uses as a child algorithm in release 3.6. There may be more uses in earlier releases.

closed 2021-03-02 by stale[bot]