mantidproject/mantid #14603: No error produced in Instrument view with badly formatted numbers

ElliotAOram, 2015-11-26:
It appears that when using 'Show Instrument` to open up the instrument view it is very forgiving of badly formatted strings in the IDF. For example I had an IDF which contained radius values in the format 0.1234.567 (containing 2 decimal points) and no error was thrown. This should be investigated.

MikeHart85, 2015-11-30:
IDF parsing is split mainly between two classes: `InstrumentDefinitionParser` and `ShapeFactory`.

Parsing of the XML is done using Poco::XML. 

`Poco::XML::Element::getAttribute()` is used to retrieve values as a `std::string`, which is then converted to the desired type using C stdlib functions `atoi()` and `atof()`. These functions are very lenient by design; they skip any leading whitespace, parse as much of the input string as  a number of the corresponding type as they can, and quietly ignore any remainder that doesn't follow the correct pattern.

So in your example of `0.1234.567`, the whole thing would be read as a string and then the `0.1234` portion would be converted into a double while the `.567` is quietly ignored.

This could be addressed by replacing all the `atoi` and `atof` calls with something that is stricter; even if it's simply a wrapper around them that does some additional validation. But this would risk that something might slip through the cracks, and future development would have to be aware of this and ensure all input is validated correctly.

A more robust solution would probably be to use XML Schema Validation. The current Schema used for instruments can be found here: http://schema.mantidproject.org/IDF/1.0/IDFSchema.xsd

This Schema describes the structure of an IDF and names of valid attributes, but does not specify any types for validation. See http://www.w3schools.com/Xml/schema_simple_attributes.asp for an example.

Unfortunately, as best as I can tell, Poco does not support schema validation anyway. The third slide in [this presentation](http://pocoproject.org/slides/170-XML.pdf) (linked from [here](http://pocoproject.org/documentation/index.html)) says:
```
valid: the document can be checked against a schema
(not supported by POCO)
```
Many other XML parsers also seem to avoid support for general purpose schema validation. One example of a library that does support it is [Xerces C++](http://xerces.apache.org/xerces-c/index.html), but it is quite heavy-weight.

We will need to decide whether we want to rework our XML handling to support Schema Validation, or just wrap existing code that reads attributes in some custom type validation checks.

NickDraper, 2016-02-19:
We have xml schema validation for the instrument files, and I thought it was in place as a system test.  Obviously is does not check this aspect.

Not worth additional time in light of geometry changes

closed 2016-03-15 by NickDraper