mantidproject/mantid #14616: IndirectQuickRun algorithm

ElliotAOram, 2015-11-27:
The MSG have asked for an algorithm that does the following:
* Loads in `.raw` or `.nxs` files
  * if files are `.raw` reduce them using data reduction
* Runs the files through the `Elwin` algorithm 
* Optional run through `Iqt`

The purpose of this is to get an overview of data quickly and in an automated way.

SpencerHowells, 2015-11-27:
Iqt is not a 'black-box', it needs thought about the input. If _red workspaces are created, just run Iqt.

SpencerHowells, 2015-12-01:
Not at all happy with Iqt option! Iqt comes with a health warning - need to be careful with number of points, energy range and energy bin otherwise the result can be rubbish!! Itr is not a black-box and should NOT be automatic.

ElliotAOram, 2015-12-01:
@SpencerHowells Probably best to exclude `I(Q,t)` from the algorithm completely in this case then.

ElliotAOram, 2016-05-20:
This will also resolve #12270

ElliotAOram, 2016-08-17:
I've commited the latest version of the algorithm (hopefully it should only require a test and documentation)

SpencerHowells, 2016-09-06:
The algo name is too vague. Can you change it to IndirectEnergyWindowScan.

louisemccann, 2016-10-24:
The EnergyWindowScan depends on a slightly modified version of MSDFit. This modified version outputs in a slightly different way, so plotting will need to be refactored.

PR #17867 (2016-11-29): Indirect - QuickRun Algorithm
MSG have asked for an EnergyWindowScan algorithm that reduces .raw files and then runs them through ElasticWindowMultiple and then an option for MSDFit. The aim of this is to speed up the workflow of Indirect algorithms. Also in this algorithm the Elastic Incoherent Structure Factor is calculated from the elastic and inelastic ranges.
EnergyWindowScan is a general algorithm that requires all the input parameters for EnergyTransfer, IndirectEnergyWindowScan is a bit more customised, this is the basis for implementation of an interface.

**To test:**

Algorithms > E [... cut]
closed 2016-11-29 by martyngigg