mantidproject/mantid #14621: LETReductionEvent2015Multirep system test failing master_systemtests-win7

FedeMPouzols, 2015-11-27:
For the last couple of nights this system test has been failing on win7 (but not any other platform) so the windows nightly builds are not being published.

Example [failure log from last night](http://builds.mantidproject.org/view/All/job/master_systemtests-win7/55/testReport/junit/SystemTests/ISISDirectInelastic/LETReductionEvent2015Multirep/), where it says at the end:
```
LETReductionEvent2015Multirep

Workspace LETreducedEi3.4 not equal to its reference file

LETReductionEvent2015Multirep

Workspace LETreducedEi8.0 not equal to its reference file
```

@abuts : I think that you know well this system test and underlying algorithms. Do you have any idea of what could be the issue? Maybe this regression comes from this recent PR: https://github.com/mantidproject/mantid/pull/14580?

abuts, 2015-11-28:
Hi, Frederico.

I will investigate the issue. The ref file for this test has been recently changed. I would suspect problem with updating files from lfs ftp

Regards,
Alex

<div>-------- Original message --------</div><div>From: Federico Montesino Pouzols <notifications@github.com> </div><div>Date:27/11/2015  21:50  (GMT+00:00) </div><div>To: mantidproject/mantid <mantid@noreply.github.com> </div><div>Cc: abuts <Alex.Buts@stfc.ac.uk> </div><div>Subject: [mantid] LETReductionEvent2015Multirep system test failing  master_systemtests-win7 (#14621) </div><div>
</div>For the last couple of nights this system test has been failing on win7 (but not any other platform) so the windows nightly builds are not being published.

Example failure log from last night, where it says at the end:

LETReductionEvent2015Multirep

Workspace LETreducedEi3.4 not equal to its reference file

LETReductionEvent2015Multirep

Workspace LETreducedEi8.0 not equal to its reference file
@abuts : I think that you know well this system test and underlying algorithms. Do you have any idea of what could be the issue? Maybe this regression comes from this recent PR: #14580?

FedeMPouzols, 2015-12-03:
This issue has been happily fixed, so putting it as "review required".

abuts, 2015-12-03:
The issue has been identified and fixed in #14644 
The follow-up ticket #14645 highlighting deeper problem has also been identified on the basis of this issue. As the difference produced due to #14645 on LET is less then 1e-3, it does not have physical significance so the #14645 is assigned and left for later.

closed 2016-01-05 by NickDraper