mantidproject/mantid #14641: Weak Pointer issue for WSGroup in Python when not in the ADS

picatess, 2015-12-01:
Acessing the children of a WorkspaceGroup is done via returning a Weak Poitner in Python. This causes issues in the scenario below
```python
# Create Group Workspace file
ws1 = CreateSampleWorkspace(WorkspaceType="Event")
ws2 = CreateSampleWorkspace(WorkspaceType="Event")
ws_group = GroupWorkspaces([ws1, ws2])
file_path = "C:/Users/pica/Desktop/group_test.nxs"
SaveNexus(InputWorkspace=ws_group, Filename=file_path)
DeleteWorkspace(ws_group)

def get_ws(file_path):
    out_name = "out_ws"
    alg_load = AlgorithmManager.createUnmanaged("LoadNexusProcessed")
    alg_load.initialize()
    alg_load.setChild(True)
    alg_load.setProperty("Filename", file_path)
    alg_load.setProperty("OutputWorkspace", out_name)
    #alg_load.setProperty("EntryNumber", 1)
    alg_load.execute()
    ws = alg_load.getProperty("OutputWorkspace").value
    ws0 = ws[0]
    return ws0
    
new_ws = get_ws(file_path)
print new_ws.getNumberHistograms()
```
The result is an invalidated workspace reference.

MichaelWedel, 2015-12-01:
Could this be related to #14120?

martyngigg, 2015-12-01:
It is all a symptom of the original choice for the lifetime handling of the workspaces through the `simpleapi` being the way it is so I slightly disagree that it's broken as that issue says.

The goal was though that you should be able to do in Python what you can in C++, just that the `simpleapi` would make certain choices for you. That is true for the standard workspaces and you can do exactly as you would in C++ if you don't use the `simpleapi` functions. In this case however you can't use the `WorkspaceGroup`s as you would like to and can in C++.

closed 2021-03-02 by stale[bot]