mantidproject/mantid #14642: Tabulated Function Requires Named Workspace

OwenArnold, 2015-12-01:
The [TabulatedFunction](http://docs.mantidproject.org/nightly/fitfunctions/TabulatedFunction.html) takes a Workspace via a name. This means that in order to run `Fit` as a child algorithm in a `DataProcessorAlgorithm`, one must perform an ADS hack to get that function to work. It would be much better from a usage perspective if it was not a requirement that the workspace sat in the ADS prior to the algorithm execution. It also leads to the potential of a runtime issues based on naming clashes while `DataProcessorAlgorithms` are executing simultaneously.

One solution could be to pass a Workspace to fit for the purposes of using the TabulatedFunction outside of the ADS. I think the solution should involve passing the workspace as a strongly typed object.

mantid-roman, 2015-12-01:
'Workspace' is an attribute. Attributes can be int, double, string or a vector of doubles. Would it be acceptable to have two vector attributes instead, one for the x and the other for the y values?

OwenArnold, 2015-12-01:
@mantid-roman Yes. I think this would be better. I would keep the current behaviour and add the new behaviour.

PR #14891 (2017-01-25): X and Y attributes for tabulated function.
Fixes #14642.

New X and Y vector attributes is an additional source of data for the TabulatedFunction.
To test try using the function with the new attributes.
closed 2017-01-25 by martyngigg