mantidproject/mantid #14659: Latex equations not generated in OS X Qt Help

martyngigg, 2015-12-02:
The current nightly build for OS X now shows the Qt help documentation as expected but the equations have not been parsed and are left as the raw latex code. See image below.

![osx-latex-error](https://cloud.githubusercontent.com/assets/733773/11526437/653622fe-98d2-11e5-8433-5998c616f6c2.png)

No such problems are seen on Windows or Linux.

quantumsteve, 2015-12-02:
I tried building from source on my office machine and equations are parsed (see below). I assume this is an issue with either the build server configuration or (unlikely) the OS X build script. I'm rerunning master-clean on `ornl-mavericks`, which I have access to and confirmed builder can run pdflatex.

[image]

quantumsteve, 2015-12-02:
Should be include a check that `PDFLATEX_COMPILER` is defined?

https://cmake.org/cmake/help/v3.0/module/FindLATEX.html

martyngigg, 2015-12-03:
Unfortunately that actually hasn't fixed it. The pull request was a sensible change though so I merged it.

I downloaded the package built by `ornl-mavericks` & it still contains the raw latex code.

martyngigg, 2015-12-03:
The build log shows a message:
```
WARNING: LaTeX command 'latex' cannot be run (needed for math display), check the pngmath_latex setting
```

martyngigg, 2015-12-03:
I've confirmed that the previous night's build on `isis-ndw1170` does display the equations. On `isis-ndw1170` the `latex` command is in `/usr/texbin` but symlinked into `/usr/local/bin`. I assume the mactex installer did that.

quantumsteve, 2015-12-03:
@martyngigg I created PR #14679 in response to the warning you copied above and think it will address that issue on at least one build server. I'll update `ornl-mavericks` and try again later today.

@peterfpeterson Would you please check that mactex is installed and in the PATH on `ornl-mac`?

peterfpeterson, 2015-12-03:
I linked `/usr/texbin/latex` into `/usr/local/bin` on `ornl-mac`.

quantumsteve, 2015-12-03:
[`ornl-mavericks` now looks good.](http://builds.mantidproject.org/view/All/job/master_clean-osx/65/artifact/build/mantid-3.5.20151203.1556-Mavericks.dmg). `ornl-mac` and `isis-ndw1170` have latex symlinked into `/usr/local/bin`. I've also [added mactex](https://github.com/mantidproject/mantid/wiki/Building-Mantid-on-OS-X-10.9-&-10.10-using-clang-and-Xcode/_compare/46027b9f486a85b66096be5b9a86d9d1469f3043...82f44ea5833858e92a6cb5832471b7d449000919) to the build instructions

Are we ready to close this issue?

martyngigg, 2015-12-04:
I just checked the latest package from `ornl-mavericks` and it displays the parsed latex correctly now too.

PR #14679 (2015-12-03): Fix latex rendering of equations on OS X.
This fixes #14659. 

cron clears the `PATH` environment variable, preventing sphinx from finding the latex executable. This PR adds /usr/texbin to the `PATH` so that latex is picked up.

This fix needs to be applied to all OS X build servers.

[Release notes](http://www.mantidproject.org/index.php?title=ReleaseNotes_3_6_UI_Changes&diff=25707&oldid=25678)

testing: Code review.
closed 2015-12-04 by martyngigg