mantidproject/mantid #14660: Ubuntu: Errors on start up with installed version

martyngigg, 2015-12-02:
Running `/opt/mantidnightly/bin/MantidPlot` on the current nightly (3.5.20151130.1920) produces errors in the console:

```
> /opt/mantidnightly/bin/MantidPlot
/opt/mantidnightly/bin/MantidPlot: 30: /opt/mantidnightly/bin/MantidPlot: cannot open 1073741824: No such file
/opt/mantidnightly/bin/MantidPlot: 30: /opt/mantidnightly/bin/MantidPlot: [[: not found
```

It would seem that on Ubuntu `/bin/sh` points to `/bin/dash` and this doesn't support the `[[` operator - http://serverfault.com/a/52050.

peterfpeterson, 2015-12-02:
Would a viable solution to change to `/bin/bash`, or is that not installed?

martyngigg, 2015-12-02:
`bash` is installed and actually the default shell for a terminal. It's just that `/bin/sh` uses `dash`. I'm happy with just changing the shebang to `/bin/bash`.

peterfpeterson, 2015-12-02:
Alternatively I can can change the `if [[ ${TCM_REPORT} < 1073741824 ]]; then` to `if [ ${TCM_REPORT} -le 1073741824 ]]; then`

peterfpeterson, 2015-12-02:
The actual problem was that I was using `bash` rather than `sh`.

PR #14674 (2015-12-02): Fixing launch scripts
**To test:** Since this is a bug (only) on ubuntu, it needs to be tested there. Try both `launch_mantidplot.sh` and `mantidpython`

This does not need to be in the release notes.

Fixes #14660.
closed 2015-12-02 by martyngigg