mantidproject/mantid #14673: Data Storage Problem. Design required.

OwenArnold, 2015-12-02:
Request from Tim Charlton, but probably affects several groups.

Where can instrument scientists store shared data that does not vary run to run. For example a flood run, or a vanadium file. Instrument scientists can often get into a mess because they use hard-coded paths in scripts to point at such workspaces, and then share scripts between themselves.

ICAT would seem to be popular, but there are issues with (1) authentication. (2) not a popular approach at ISIS at least. 

I've marked this as SSC item because Tim is in the process of moving facilities and also because it will undoubtably be raised at this SSC and it would be nice to be ahead of the curve with a solution.

Jon-Taylor, 2015-12-02:
i’ve forwarded this to Tobias, i think he will have some ideas on the way forward with this
jon

zjttoefs, 2016-01-27:
I agree ICAT would be my primary idea for holding a pointer to the data (not the actual data). 
Ideally that could also be encoded in the raw data file, but that may not always work or may require an update with a later calibration run. Updating ICAT later is safer than modifying the raw data file. 

That doesn't solve the problem of actually storing the calibration data. I guess both is a problem?

NickDraper, 2016-10-14:
Not raised in a couple of years. Archived

closed 2016-10-14 by NickDraper