mantidproject/mantid #14688: Indirect Data Reduction

ElliotAOram, 2015-12-03:
Several of the interfaces in IDR are reported to be automatically running algorithms when values in the property trees are changed. This is causing issue with inputting values and hence should be removed.

*The first step is to investigate which interfaces are doing it and on what inputs.*

SpencerHowells, 2015-12-03:
Not just property tree values but sometimes changing anything eg run/file name,

ElliotAOram, 2015-12-09:
* **`ISIS Calibration'**
  * Runs `ISISIndirectEnergyTransfer` on Change of Run number
  * This appears to be for the plot mini plot in Calibration
* **`ISISDiagnostic`**
  * Runs `TimeSlice` algorithm on change of run number and parameter change
* **`Moments`**
  * Change of file and parameter cause re-run of `SofQWMoments`

PR #14733 (2015-12-11): Indirect Data Reduction Remove AutoRun in interfaces
Fixes #14688

The ISISDiagnostics and Moments interfaces should no longer auto run the main algorithms when data is changed including file changes and property changes.

# To Test 
* Change default instrument to `IRIS` at `ISIS` View> Preferences > Mantid > Facility/Default Instrument (This should not have to be done after 14607)
* Ensure `Search Data Archive` is checked in the `Manager User Directory` window

# ISIS Diagnostics
* Open ISIS Diagnostics (Interfaces > Indirect > Data Reduction > ISIS Diagnostics)
* Input Files = 26176
* Ensure [... cut]
closed 2015-12-11 by LamarMoore