mantidproject/mantid #14697: ImagingIMATTomoScripts disabled for Python 2.6

FedeMPouzols, 2015-12-04:
A first attempt (#14678) to make this system test work on rhel6 was not quite enough.
The nightly system tests failed again:
- From the unittest front, there's at least the issue that AssertRaises works as context manager only in Python >2.7.
- From the numpy front, the version available is from 2010?-update-from-2012?. numpy test methods are not supported (requires changes to the test), and more methods will need to be replaced in the code that the test tries to cover (rot, etc.).

This system test raises many issues with Python 2.6 so it's being disabled when Python <2.7.

The issue is there, but I think the effort/gain ratio doesn't justify any more time spent on this at the moment. If rhel6 is deprecated soon (hopefully) this issue can be closed.

FedeMPouzols, 2016-08-04:
@DimitarTasev: for the moment I do not think this is worth fixing. In the hopefully not so distant future when we deprecate RHEL6 (maybe in just a few months?) this will not be an issue so this GitHub issue could be closed.

I'll tell you later about the imaging system and unit tests.

DimitarTasev, 2017-01-16:
#18386

closed 2017-01-16 by DimitarTasev