mantidproject/mantid #14716: New algorithm CreateCacheFilename

peterfpeterson, 2015-12-08:
Taken from the [Workflow Caching Design](https://github.com/mantidproject/documents/blob/master/Design/WorkflowCaching.md):

The fundamental component that is missing is an "easy" way to create the hash from a set of processing parameters. This will be done as an Algorithm and borrow ideas from [python tempfile](https://docs.python.org/3/library/tempfile.html).
The algorithm will accept a prefix, [PropertyManager](http://docs.mantidproject.org/nightly/api/python/mantid/kernel/PropertyManager.html), whitelist of properties to use, blacklist of properties to exclude, and a string array (or `List`), and a directory for cache files to exist in.
All of these properties are individually optional, but failing to specify any should result in an error.
The whitelist and blacklist will be used to select which of the properties in the `PropertyManager` will be used to calculate the hash.
Both will be interpreted as globbing.
The string array is will be key/value pairs of properties that should be considered, but are not in the provided `PropertyManager`.
If a directory is not specified, cache files will go into a `cache` subdirectory of `ConfigService::getUserPropertiesDir()`.
On unix this will be `~/.mantid/cache`.

The algorithm will convert all properties to strings as `"%s=%s" % (property.name, property.valueAsStr)`, sort the list, then convert it to a `sha1`.
A filename with the form `<location>/<prefix>_<sha1>.nxs` will be returned as the output property.
If no prefix is specified then file result will be `<location>/<sha1>.nxs`.

yxqd, 2015-12-09:
Questions:
 - Are whitelist and blacklist strings of patterns to match property names?
 - Is default whitelist '*' to match everything?
 - Is default blacklist '' to match nothing?
 - What is the string array (or list) parameter used for? The output filename?

peterfpeterson, 2015-12-10:
I updated the description, but you should look at the example towards the bottom of the design document to get a better idea of usage.

yxqd, 2015-12-10:
Thanks Pete. I gather from the example script that the parameters are:

 - property_manager: an instance of PropertyManager from which property values can be retrieved. None means we don't care about property manager -- all properties will come from other_properties
 - properties: a list of strings. each string is a property managed by the given property_manager. but empty list means taking all properties from the property_manager
 - other_properties: a list of strings. each string is in the form of "<key>=<value>" for one property not managed by the property_manager
 - prefix: prefix to the output hash name
 - cache_dir: the directory in which the cach file will be created

Please comment if anything is obviously misinterpreted.

peterfpeterson, 2015-12-10:
...and the cache directory should default to `~/.mantid/cache` on linux (see design for how).

Both the prefix and cache directory should act similar to utilities for generating temporary files.

PR #14831 (2015-12-29): new alg createcachefilename
this fixes #14716
closed 2015-12-29 by peterfpeterson