mantidproject/mantid #14744: Handle cummulative sample logs correctly when adding adding.

picatess, 2015-12-10:
Some sample logs which contain time series such as `good_frames` represent a cummulative time series. Our current approach is to concatenate time series properties when they are added. 

As part of this maintenance ticket, the current functionality should be altered to take this into account. Currently we have an array of ADDABLES which gets added correctly. This list is stored in `Run.cpp`

@martyngigg suggested that we annotate each sample log when it is created in the Loaders in order to later on decide dynamically what the add-behaviour of a particular sample log would be. Later on when we want to add workspaces a correct add-policy based on the annoation is chosen. In this way we can handle our current problem and get rid of a potentially "outdated" array of ADDABLES in Run.cpp.

martyngigg, 2015-12-10:
I think this could use a small design document to make the ideas more concrete and check we haven't missed anything. I'll put something together.

picatess, 2015-12-10:
An important edge, or in fact common case occurs when a workspace is added to itself. Currently we only consider the sample log of one workspace. If we don;'t consider the sample log of a single workspace only, then it is not clear how the cumulative sample log should be defined for this case. This is also true if the workspaces are not identical, but overlap in their TimerSeriesProperty.

closed 2021-03-03 by stale[bot]