mantidproject/mantid #14745: ExponentialMD does not carry out the expected calculation

matthew-d-jones, 2015-12-10:
Problem reported by Russell Ewigs.
Check where ExponentialMD is used and make sure there is a consensus that it should do as Russell suggests.

```python
#Exp, and other unary ops
w_exp=ExponentialMD(out_md_2d)
sv=plotSlice(w_exp,colormin=0.01,colormax=1,normalization=2)
#Unexpected behaviour - would expect to see plot of exp(s/n-events), not exp(sig)
/ n-events
```

AndreiSavici, 2015-12-10:
What about the case where you don't want number of event normalization? If you have an MD workspace produced from an event workspace (say diffraction at SNS), the number of event normalization would produce a histogram with 1 everywhere where you have at least 1 count, and nans everywhere else. It would be very boring image as soon as I use this normalization

matthew-d-jones, 2015-12-11:
I'm closing this issue as it is really part of a wider question around MDWorkspaces which is already under consideration and discussion.

closed 2015-12-11 by matthew-d-jones