mantidproject/mantid #14755: Check getcycle.py file in PowderISIS

shahrozahmed1, 2015-12-11:
There is not a lot going on inside the `getcycle.py` file right now, it would be ideal to check with Ron whether they required this file for anything particular. If yes then it needs to be documented what the file does and how. If the file is not necessary any more it can be removed, as it is not required for the diffraction process.

DavidFair, 2016-09-16:
@FedeMPouzols This script seems to do something similar to using the "Search Data Archive" option - as noted in the issue I don't believe we use it anywhere so could it be removed?

FedeMPouzols, 2016-09-16:
Yes, I think it is safe to remove this, as nobody has requested it and it uses the old manual way of finding files from the archive.

PR #17519 (2016-09-16): Removed unused python script from CryPowderISIS
Description of work.

This PR removes an unused original powder diffraction script from the powder diffraction group. It performed the manual method of retrieving files from the data archive which Mantid handles itself.

**To test:**
If build passes it can be shipped

Fixes #14755 .

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the co [... cut]
closed 2016-09-16 by FedeMPouzols