mantidproject/mantid #14782: Set masked data in MDWorkspaces to zero

matthew-d-jones, 2015-12-14:
Currently you can set masks in MDWorkspaces without loosing any data. You can therefore later remove or change the mask. However, it was decided in the VATES meeting 14 Dec 2015 that this does not need to be the case.

Set masked signal and errors to zero. The advantages of this are that it is consistent with MatrixWorkspaces and that many algorithms will simply "do the right thing" without us having to write code to handle masked data.

AndreiSavici, 2015-12-14:
I would use NaNs instead. SliceViewer uses transparent color for NaN

matthew-d-jones, 2015-12-15:
NaNs display as white in the SliceViewer, which we want to avoid as it is on the colourmap. If it really counts as transparent then maybe there is a way to set the background colour to something other than white? In principle I also agree that NaNs would be preferable though.
I think MatrixWorkspaces use 0 for masked data so that many algorithms treat masked data in the correct way without having to explicitly code the behaviour. But perhaps the same would also be true if NaNs were used.
I will check all of these things before designing a solution and try to make sure that the solution is common across workspace types.

PR #14937 (2016-01-14): Set masked data to NaN in MDWorkspaces
Closes #14782 

In the case of MDEventWorkspaces this means setting the signal and error to NaN for masked boxes, the actual events are not removed. This change should mean that displaying masked MDWorkspaces in the VSI and Slice Viewer. Masked regions should be invisible in the VSI and match out-of-bounds of the workspace in the Slice Viewer (white).

NaN displays as a high number in 1D plots, which is undesirable for masked data, but this will be addressed in #14781.

**To Test**

You can create a masked MDEventWorkspace and MDHistoWorkspace with the fo [... cut]
closed 2016-01-14 by picatess