mantidproject/mantid #14787: Better Model For Stitching in Reduction

OwenArnold, 2015-12-15:
This is the procedure that Mario wants to be able to use for SURF reduction. This works well for him.
**The objective is to get both Reflectometry GUIs able to produce stitched outputs in the same manner**.

1. Notice that in Stitch1D **ScaleRHSWorkspace=False** is being applied. We don't want to change the defaults in the current one, but we do need a switch for this somehow.
2. The Q-range as calculated from the tables are not good enough to give a good overlap. That's why they use the script below. The suggestion is that somehow The calculated Q-Range displayed in the table is a truncated version of the full Q-Range. This needs investigation.

```python
##Make the transmission:

Load(Filename=r'\\isis\inst$\ndxsurf\instrument\data\cycle_15_3\SURF00114000.nxs', OutputWorkspace='_114000')
CreateTransmissionWorkspaceAuto('_114000', OutputWorkspace='TRANS', Params='0.05', WavelengthMin='0.5', WavelengthMax='7')

######
##Set the save directory:
#(You'll need to change this)
config['defaultsave.directory'] = 'C:\Users\yzo13419\Documents\User Data\Clarke - 1520071\Surf Data'

##Set instrument if not already SURF:
config['default.instrument'] = 'SURF'

#############
#############
##Reduction - this assumes 3 runs in a row at 0.35, 0.75 and 1.5 but can obviously be tweeked:

def ReflHighStitch(Run1, Min1=0.011,Max1=0.038, Min2=0.026, Max2=0.08, Min3=0.054, Max3=0.2):  #Assumes 3 angles in a row.
    x1=Run1
    x2=x1+1
    x3=x2+1
    Load(Filename=r'\\isis\inst$\ndxsurf\instrument\data\cycle_15_3\SURF00'+str(x1)+'.nxs', OutputWorkspace=str(x1))
    Load(Filename=r'\\isis\inst$\ndxsurf\instrument\data\cycle_15_3\SURF00'+str(x2)+'.nxs', OutputWorkspace=str(x2))
    Load(Filename=r'\\isis\inst$\ndxsurf\instrument\data\cycle_15_3\SURF00'+str(x3)+'.nxs', OutputWorkspace=str(x3))
    
    ReflectometryReductionOneAuto(InputWorkspace=str(x1), FirstTransmissionRun='TRANS', OutputWorkspace=str(x1)+'_IvsQ', OutputWorkspaceWavelength=str(x1)+'_IvsLam', ThetaIn=0.35, WavelengthMin=0.5, WavelengthMax=7)
    Rebin(InputWorkspace=str(x1)+'_IvsQ', Params=str(Min1)+',-0.03,'+str(Max1), OutputWorkspace=str(x1)+'_IvsQ')
    ReflectometryReductionOneAuto(InputWorkspace=str(x2), FirstTransmissionRun='TRANS', OutputWorkspace=str(x2)+'_IvsQ', OutputWorkspaceWavelength=str(x2)+'_IvsLam', ThetaIn=0.75,WavelengthMin=0.5, WavelengthMax=7)
    Rebin(InputWorkspace=str(x2)+'_IvsQ', Params=str(Min2)+',-0.03,'+str(Max2), OutputWorkspace=str(x2)+'_IvsQ')
    ReflectometryReductionOneAuto(InputWorkspace=str(x3), FirstTransmissionRun='TRANS', OutputWorkspace=str(x3)+'_IvsQ', OutputWorkspaceWavelength=str(x3)+'_IvsLam', ThetaIn=1.50,WavelengthMin=0.5, WavelengthMax=7)
    Rebin(InputWorkspace=str(x3)+'_IvsQ', Params=str(Min3)+',-0.03,'+str(Max3), OutputWorkspace=str(x3)+'_IvsQ')
    
    CloneWorkspace(InputWorkspace=str(x1)+'_IvsQ', OutputWorkspace='currentSum')
    currentSum, scale1 = Stitch1D(LHSWorkspace='currentSum', RHSWorkspace=str(x2)+'_IvsQ', StartOverlap=0.025, EndOverlap=0.051, Params='0.011089,-0.03,0.2', ScaleRHSWorkspace=False)
    currentSum, scale2 = Stitch1D(LHSWorkspace='currentSum', RHSWorkspace=str(x3)+'_IvsQ', StartOverlap=0.050, EndOverlap=0.11, Params='0.011089,-0.03,0.2', ScaleRHSWorkspace=False)
    RenameWorkspace(InputWorkspace='currentSum', OutputWorkspace=str(x1)+'_'+str(x3))
    
    Scale(InputWorkspace=str(x1)+'_IvsQ', OutputWorkspace=str(x1)+'_IvsQ', Factor=scale1*scale2)
    Scale(InputWorkspace=str(x2)+'_IvsQ', OutputWorkspace=str(x2)+'_IvsQ', Factor=scale2)
    print scale1, scale2
    
    plotSpectrum(mtd[str(x1)+'_'+str(x3)], 0, error_bars=True)
    plotSpectrum((mtd[str(x1)+'_IvsQ'], mtd[str(x2)+'_IvsQ'], mtd[str(x3)+'_IvsQ']), 0, error_bars=True) #Plot all three angles too to show the stitching regions...
    
    ##Allows the generated file to be saved out:
    SaveAscii(InputWorkspace=str(x1)+'_'+str(x3), Filename=str(x1)+'_'+str(x3)+'.dat', Separator='Tab', WriteSpectrumID=False)
    ##OR, brings up the pop-up box to enable you to choose to save it or not:
    #SaveAsciiDialog(InputWorkspace=str(x1)+'_'+str(x3), Filename=str(x1)+'_'+str(x3)+'.dat', Separator='Tab', WriteSpectrumID=False)
    
##################
##################

##Example use either with or without specifying the ranges for each angle:
ReflHighStitch(113976)
ReflHighStitch(113976, Min1=0.011,Max1=0.038, Min2=0.026, Max2=0.08, Min3=0.054, Max3=0.2)
```

MattKing06, 2015-12-17:
Below is a table containing the values for the `Params` property in the Rebin part of the reduction.
The Rebinning of workspaces in our algorithm is run after the reduction using the {Qmin, dQ/Q, Qmax} as the params. Whereas the values for `Params` in the script version above uses hard coded values passed to the method. 

This table is to demonstrate the relationship between the Params that we find in our algorithm and the hard-coded values in the script above (if there is any relationship to demonstrate).

| PARAM  | ALGORITHM VALUE  | SCRIPT VALUE | FACTOR (script/alg)|
| ---------- | ------------ | ------------- | -----------------|
| `Q1:min `| 0.005486 | 0.011  |2.01|
| `Q1:dQ/Q` | -0.03  | -0.03 | 1.0 |
|  `Q1:max` | 0.0754938|0.038 |0.503|
| `Q2:min `| 0.0117563 | 0.026  |2.21|
| `Q2:dQ/Q` | -0.03  | -0.03 |1.0|
|  `Q2:max` | 0.161771|0.08 |0.495|
| `Q3:min `| 0.02351 | 0.054  |2.30 |
| `Q3:dQ/Q` | -0.03  | -0.03 |1.0|
|  `Q3:max` | 0.323536|0.2 |0.618|

MattKing06, 2016-01-05:
After running just ReflectometryReductionOne on `SURF113976`the values of Qmin and Qmax are the same as the results I get after running just this part of the script:

```python
##Make the transmission:

Load(Filename=r'\\isis\inst$\ndxsurf\instrument\data\cycle_15_3\SURF00114000.nxs', OutputWorkspace='_114000')
CreateTransmissionWorkspaceAuto('_114000', OutputWorkspace='TRANS', Params='0.05', WavelengthMin='0.5', WavelengthMax='7')

######
##Set the save directory:
#(You'll need to change this)
config['defaultsave.directory'] = 'C:\Users\yzo13419\Documents\User Data\Clarke - 1520071\Surf Data'

##Set instrument if not already SURF:
config['default.instrument'] = 'SURF'

#############
#############
##Reduction - this assumes 3 runs in a row at 0.35, 0.75 and 1.5 but can obviously be tweeked:

def ReflHighStitch(Run1, Min1=0.011,Max1=0.038, Min2=0.026, Max2=0.08, Min3=0.054, Max3=0.2):  #Assumes 3 angles in a row.
    x1=Run1
    x2=x1+1
    x3=x2+1
    Load(Filename=r'\\isis\inst$\ndxsurf\instrument\data\cycle_15_3\SURF00'+str(x1)+'.nxs', OutputWorkspace=str(x1))
    Load(Filename=r'\\isis\inst$\ndxsurf\instrument\data\cycle_15_3\SURF00'+str(x2)+'.nxs', OutputWorkspace=str(x2))
    Load(Filename=r'\\isis\inst$\ndxsurf\instrument\data\cycle_15_3\SURF00'+str(x3)+'.nxs', OutputWorkspace=str(x3))

    ReflectometryReductionOneAuto(InputWorkspace=str(x1), FirstTransmissionRun='TRANS', OutputWorkspace=str(x1)+'_IvsQ', OutputWorkspaceWavelength=str(x1)+'_IvsLam', ThetaIn=0.35, WavelengthMin=0.5, WavelengthMax=7)
```

The difference in Qmin and Qmax values only appears after this line in the script:

```python
Rebin(InputWorkspace=str(x1)+'_IvsQ', Params=str(Min1)+',-0.03,'+str(Max1), OutputWorkspace=str(x1)+'_IvsQ')
```

When we use the hardcoded values `0.011` and `0.038` as our min and max in the script. 
We need to find out why these values are being used, is it do with having more interest with certain spectra range than others, or is there an actual relationship between the Qmin and Qmax of the reduced workspace and the min and max that the script is rebinning to?

As I see it, if the workspaces are not the same after the reduction and rebinning, then it is unlikely they will be similar when it comes to stitching them.

MattKing06, 2016-01-05:
Why must we calculate the qrange [here](https://github.com/mantidproject/mantid/blob/14702_reduce_pylint_warnings_for_reflectometer_files/MantidQt/CustomInterfaces/src/Reflectometry/ReflMainViewPresenter.cpp#L766) by passing in the IvsQ workspace? is there a reason why we can't just read the min and max straight from the workspace?

OwenArnold, 2016-01-05:
What you are saying implies there is a problem here https://github.com/mantidproject/mantid/blob/14702_reduce_pylint_warnings_for_reflectometer_files/MantidQt/CustomInterfaces/src/Reflectometry/ReflMainViewPresenter.cpp#L843:L844 vs what ConvertUnits is doing. That could be worth investigating.

MattKing06, 2016-01-05:
I agree that this a problem, but it still doesn't help determine why the hard-coded values are used as they still won't match up with what we are seeing from our version of the algorithm. However, I will investigate this as it is bad to duplicate this process when it's not necessary.

closed 2021-03-01 by stale[bot]