mantidproject/mantid #14848: Upgrade Reflectometry Reduction (Polref) Interface for Scaling of LHS Workspace

OwenArnold, 2015-12-23:
See the original PR and Issue #14847, whereby a fix has been made to the old UI.

We would like to do the same thing to the new UI. The nice way to do this would be to allow the user to provide the override in the options column, however, the options column is only for the ReflectometryReductionOneAuto properties. Stitching is an optional post-processing step. We need to adapt the UI so that the options column also includes any of the options used for stitching. That way users can specify *ScaleRHSWorkspace=False* in the options column and the LHS workspace will be scaled instead.

MattKing06, 2016-03-03:
After some investigation it seems necessary to create a new type of `AlgorithmHintStrategy` that can take 2 or more algorithms and a list of their properties so that the options of both `Stitch1DMany` and `ReflectometryReductionOneAuto` can be overridden accordingly. Do you have any opinions on this @OwenArnold?

OwenArnold, 2016-03-03:
@MattKing06 I think you should talk to @raquelalvarezbanos about this because she's just about to start doing something similar with CreateTransmissionWorkspace for the same purposes. However, I think that the best course of action is not to adapt the Options column now. I think we should do something very similar to what was done for Max's polref interface. I know that I said originally that the options column should be modified, but I now think this is not a good way to do things.

raquelalvarezbanos, 2016-03-10:
@MattKing06 When you are back, before you continue working on this, please remind me that we have to talk about a couple of things related to this issue (unit tests and stitching options).

MattKing06, 2016-06-17:
We now have `Pre-processing` and `Post-processing` steps as a part of the new DataProcessor UI. The `post-processing` steps now include `Stitch1DMany` and all the properties of this algorithm are available to set via the `post-processing` text box.

closed 2016-10-03 by NickDraper