mantidproject/mantid #14849: Create a flat projection surface for the Instrument View

mantid-roman, 2015-12-23:
The instrument is to be projected onto one of the planes: XY, XZ, or YZ.

AndreiSavici, 2015-12-23:
Is this the same as http://docs.mantidproject.org/nightly/algorithms/ConvertAxesToRealSpace-v1.html ?

mantid-roman, 2015-12-24:
No it is for the instrument view. It won't do rebinning, but allow detector picking and selection. It is similar to the spherical and cylindrical projections that we have there.

OwenArnold, 2015-12-24:
This should be easy enough if the panels do not overlap. If there is overlap, do you think the view mode could try to at least keep the panels relative to each other in some way.

mantid-roman, 2015-12-24:
Maybe it shouldn't be used for such instruments?

OwenArnold, 2015-12-24:
Agreed, wouldn't make much sense for a spherical geometry. But for lots of SANS instruments it definitely makes sense and overlaps can occur.

closed 2021-03-02 by stale[bot]