mantidproject/mantid #14861: Umbrella: Add workflow diagrams to algorithm documentation

LamarMoore, 2016-01-04:
Add workflow diagrams to the following algorithms

 - [ ] ComputeSensitivity
 - [ ] CreateTransmissionWorkspaceAuto
 - [ ] SofQW
 - [ ] AccumulateMD
 - [ ] CreateMD
 - [ ] CutMD
 - [ ] SANSBeamFluxCorrection
 - [ ] SANSStitch
 - [ ] TOSCABankCorrection
 - [ ] IndirectFlatPlateAbsorption
 - [ ] IndirectCylinderAbsorption
 - [ ] IndirectAnnulusAbsorption
 - [ ] EVSDiffractionReduction
 - [ ] DetectorFloodWeighting
 - [ ] SNSPowderReduction
 - [ ] MuscatSofQW

OwenArnold, 2016-01-04:
CreateMD -> Matthew D Jones
AccumulateMD -> Matthew D Jones
CutMD -> Matthew D Jones 

CreateTransmissionWorkspaceAuto -> Owen Arnold
DetectorFloodWeighting -> Owen Arnold

ElliotAOram, 2016-01-04:
IndirectFlatPlateAbsorpton -> ElliotAOram
IndirectCylinderAbsorption -> ElliotAOram
IndirectAnnulusAbsorption -> ElliotAOram
EVSDiffractionReduction -> ElliotAOram

NickDraper, 2016-01-04:
SNSPowderReduction -> Peter Peterson

closed 2021-03-03 by stale[bot]