mantidproject/mantid #14883: Q-E Coverage graphs for direct geometry instruments

mducle, 2016-01-05:
To aid planning experiments it would be useful to have plots of the detector trajectories (in Q-E space) for a given incident energy for a given instrument. Assuming powder averaging, this would require only knowledge of the minimum and maximum detector angles (and any detector gaps).

We'd like to contribute a python algorithm which does this calculation and produces a workspace with the required Q- and E-data and plots it.

stuartcampbell, 2016-01-05:
I notice that this is only for the ISIS instruments, can it be made generic please ?

stuartcampbell, 2016-01-05:
@AndreiSavici or @yxqd can help with it for the SNS

mducle, 2016-01-05:
Sure! I just had the info for the ISIS detectors available more readily. The necessary info is in the instrument definition files, so I'll just look them up now... I was just lazy I guess.

stuartcampbell, 2016-01-05:
Excellent - Thank you.

SpencerHowells, 2016-01-06:
Would also be of use to indirect geometry instruments especially for large energy transfers.

NickDraper, 2016-04-04:
resolved by #14997

closed 2016-04-04 by NickDraper