mantidproject/mantid #14890: Abstract Instrument View from MantidPlot

LamarMoore, 2016-01-06:
The ESS team requested abstracting the Instrument View tool from the main MantidPlot code base with the hope of eventually using this as a stand alone executable.

This is currently under investigation.

SimonHeybrock, 2016-01-07:
Alternatively, if not as a stand alone executable, having it as a widget that can be included in other Qt applications might be useful.

Also consider making it available from within Python, i.e., when importing the Mantid Python API, what is missing that keeps us from using/starting the Instrument View? Maybe this is the easiest starting point (and it would get us a long way)?

MichaelWedel, 2016-01-11:
Before I forget it, the framework I saw being used to expose C++ code to python is called [Swig](http://www.swig.org/). I have never used it though, so I can not comment whether it's easier than any of the other options (sip, boost::python).

martyngigg, 2020-10-09:
This is now possible from both C++ & Python without requiring MantidPlot or MantidWorkbench. The code is in a separate library with a Python wrapper inside the `mantidqt` package.

closed 2020-10-09 by martyngigg