mantidproject/mantid #14925: Unreliable unit test - ConvertEventsToMDTest

martyngigg, 2016-01-11:
`ConvertEventsToMDTest` has been persistently failing (899!= 900) in a small percentage of runs (say 5-10%?) for some time now. Only on windows (debug and release). I'd suspect this started with the upgrade to MSVC2015.  The issue could be a false positive or a glitch in the test and not the tested code. The test generates the 900 events with `rand()`. Someone that understands well that test and how the random integers are mapped into Q space: could  a poor `rand()` be generating events outside of the normal expected cube, or duplicate/invalid events?

martyngigg, 2016-01-11:
It has been suggested that `rand()` is not threadsafe and could be the cause of the random failures.

As a modernization task  `rand()` probably should be changed to instead use <random>.

FedeMPouzols, 2016-01-11:
Sorry I missed this. There's an older duplicate of this issue: #14556.

martyngigg, 2016-01-11:
My fault, I missed it. I'll close this as a duplicate.

FedeMPouzols, 2016-01-11:
If I remember correctly, the test also fails single-threaded runs if you run it in a loop enough times. I'll check if the modernization fixes the issue.

PR #14950 (2016-01-13): Modernize rand() => Kernel::MersenneTwister in WorkspaceCreationHelper to "fix" ConvertEventsToMDTest
Fixes #14556.

This simply replaces calls to rand() in the `WorkspaceCreationHelper` file with `Kernel::MersenneTwister`. This seems to be enough to indirectly fix the issue with the test `ConvertEventsToMDTest` (which failed now in last night's clean win7 build!).

Normally it would take 5-30 repetitions on windows to get the failure with rand(). It remains unknown what pattern of tof-pulsetime random values rand() generates in MSVC that makes the test fail. It is not about duplicate pairs no [... cut]
closed 2016-01-13 by NickDraper