mantidproject/mantid #14944: Kernel::MersenneTwister: replace the uncerlying boost implementation with the C++11 one

FedeMPouzols, 2016-01-12:
Now that we have C++11 support on all platforms it should be possible to just replace `<boost/random/mersenne_twister.hpp>` and related headers, using the `std:mt19937` implementation available from  `<random>`. 

Remove the dependency on `boost/random/` headers. The `std` implementation resembles the `boost` one very closely so it should be possible to do this as a one-to-one replacement.

FedeMPouzols, 2016-07-14:
@NickDraper: this is in principle simple and could be good practice for new starters.

NickDraper, 2016-07-14:
I suggest we keep this one for maintenance

PranavBahuguna, 2016-09-26:
While all the boost versions of the components of the Mersenne Twister algorithm could be replaced easily with the C++11 versions, this swap has caused several algorithm unit tests to fail (e.g. MayersSampleCorrectionTest, MonteCarloAbsorptionTest etc.). In tests such as `MayersSampleCorrectionTest::test_Success_With_Both_Corrections()`, I did notice that the actual results are still quite close to the expected ones, so I suspect that the precision of the random numbers generated from the new Mersenne Twister implementation is lower than that of the old implementation. What should I do about this?

gemmaguest, 2020-10-29:
Closing as stale because we are unlikely to get around to this

closed 2020-10-29 by gemmaguest