mantidproject/mantid #14952: Possible EISF miscalculation in ConvFit

ElliotAOram, 2016-01-13:
>Franz has pointed out a possible error in this calculation.
The elastic is done using the DeltaFunction algo which uses Height to scale the resolution.
So the elastic INTENSITY is Height times the Res intensity (area) and we are using just HEIGHT.
Thus we need to integrate the Res & multiply by Height to give elastic intensity.

This bug may well be in ConvolutionFitSequential. This will have to be investigated.

SpencerHowells, 2016-02-24:
The DeltaFunction is Height*Res(x), so its intensity is Integral(Height*Res(x)) = Height*Integral(Res).
The Lorenztian is A*Lor(x) where Lor(x) integrates to 1. Its intensity is Integral(Res(x)*A*Lor(x)) = Integral(Res)*Integral(A*Lor) = A*Integral(Res).
So using Height and A for the EISF is correct.

ElliotAOram, 2016-02-24:
Based on Spencer's comment on this issue, we believe that EISF is in fact calculated correctly a such this issue can be closed

closed 2016-02-24 by ElliotAOram