mantidproject/mantid #14970: Symmetry operations bugs

AndreiSavici, 2016-01-14:
I am trying to use symmetry operations to transform some coordinates, to use them in symmetrization algorithms
```
symOp = SymmetryOperationFactory.createSymOp("x,y,z+1")
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-1-103640a8447c> in <module>()
----> 1 symOp = SymmetryOperationFactory.createSymOp("x,y,z+1")

RuntimeError: Error in parsing symbol x,y,z+1:
Failed to parse component string z+1: Could not parse entire string. in  on line 0

symOp = SymmetryOperationFactory.createSymOp("x,y,z+10/20")
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-2-d00995784753> in <module>()
----> 1 symOp = SymmetryOperationFactory.createSymOp("x,y,z+10/20")

RuntimeError: Error in parsing symbol x,y,z+10/20:
Failed to parse component string z+10/20: Could not parse entire string. in  on line 0

symOp = SymmetryOperationFactory.createSymOp("x,y,z+5/3")
print symOp.transformCoordinates([1.,2.,3.])

[1,2,3.66667]
```

Here are the things that need to be fixed:
1. Be able to add an integer 1 or 2 to one of the components
2. Be able to use more than one digit
3. Need to have 5/3 be 1.6667, not 0.6667

MichaelWedel, 2016-01-15:
I think all of these might be gone with PR https://github.com/mantidproject/mantid/pull/14913. I will verify this once I'm at my workstation.

MichaelWedel, 2016-01-15:
Okay, all of these are parsed in the branch with the new parser, but the modulo operation is still in place. I'll try to see what happens if it's removed.

MichaelWedel, 2016-01-15:
It is possible to move the modulo operation to where it's needed for `Group` to work properly (`operator*(const SymmetryOperation &other`). That solves most of the problem. But somehow I'm not entirely satisfied with this solution, all this modulo-stuff should happen strictly within `Group` where it's actually required, so I need to look at this a bit more.

PR #15455 (2016-02-29): Enable SymmetryOperation to have vector-components >= 1
Fixes #14970.

To enable the requested behavior it seemed appropriate to do some minor cleanup in `SymmetryOperation` first, the creation of actual `std::pair<IntMatrix, V3R>` instances was dropped for some intermediate steps during parsing/creating in favor of `MatrixVectorPair<int, V3R>`.

The procedures for wrapping vector components to the appropriate interval are now part of `Group` where they are explicitly carried out on the symmetry operations.

Nevertheless, all of this is more or less internal and I don't think this needs to go  [... cut]
closed 2016-02-29 by martyngigg