mantidproject/mantid #15014: Overview user documentation for fitting minimizer

Anders-Markvardsen, 2016-01-18:
At present the document for what minimizer a user may chose (other than the default) on the page http://docs.mantidproject.org/nightly/algorithms/Fit-v1.html is limitted to:

```
Using the Minimizer property, Fit can be set to use different algorithms to perform the minimization. By default if the function’s derivatives can be evaluated then Fit uses the GSL Levenberg-Marquardt minimizer.
``` 

Add a new section to http://docs.mantidproject.org/nightly/algorithms/Fit-v1.htm, for instance, call it 'Choices of minimizers', where a high level summary is given of the different minimizers we support and for which problems a user may benefit from using different minimizers. For example that for very noisy data that user may benefit from using a minimizer different from Levenberg-Marquardt and so on.

Anders-Markvardsen, 2016-07-01:
This could also be a challenging new starter ticket

mantid-roman, 2016-07-01:
I am converting some top level `tex` doc from `ral_nlls` library to `rst` to see how hard it is.

PR #17960 (2016-11-22): Basic documents of fitting function minimizers
A new folder has been created for fitting function minimizer documents and some short documents for minimizers have been added

**To test:**

Build the HTML documents. The go to the FittingMinimizers in Concepts. Verify that seven of the minimizers list there link to short documents in a new folder for fitting function minimizers.
Check that each of these documents links to a Wikipedia page describing the algorithm.
Also check that all links are OK.

Build QTHELP and test the Mantid help for the information already described.

Fixes #15014 .

---
 [... cut]
closed 2016-11-22 by martyngigg