mantidproject/mantid #15028: Use CalculateChiSquared alg for tests in WeightedLeastSquaresTest.py

Anders-Markvardsen, 2016-01-19:
Use something like

```
CalculateChiSquared('name=Gaussian,Height=0.0994414,PeakCentre=0.000375018,Sigma=4.96144e-06;name=Gaussian,Height=0.121684,PeakCentre=0.000165893,Sigma=2.13907e-05','Slice',WorkspaceIndex=0,Output='res')
```

to test even for what appear to be good fitting cases that the minimum reached for tests in WeightedLeastSquaresTest.py.have reach a local minimum well.

Add line towards top of WeightedLeastSquaresTest.py to recommend using CalculateChiSquared to check if 'true' minimum has been found for when adding new data examples.

@mantid-roman

FedeMPouzols, 2016-02-01:
Current status: more discussions with SCD numerical analysis group would be advisable before going ahead with this.

FedeMPouzols, 2016-09-22:
I think this issue needs revisiting after the developments in the last few months.
@tom-perkins, @Anders-Markvardsen

closed 2021-03-01 by stale[bot]