mantidproject/mantid #15066: Speed up CalMuonDetectorPhases

raquelalvarezbanos, 2016-01-22:
`CalMuonDetectorPhases` fits simultaneously the histograms in a muon dataset. Some parameters are shared (the frequency of the oscillations) and some are independent (the amplitude and phase). The algorithm creates a multi domain function to fit the data. However, muon scientists usually want to fit ~96 histograms, and this has been reported to take >7 min. This issue is to try to speed up the algorithm.

raquelalvarezbanos, 2016-01-22:
Comment from James:

> One could think how to optimise CalMuonDetectorPhases, for example Phase 1: Group the workspace (choosing an appropriate group for the field direction being used), do AsymmetryCalc, and fit to an oscillating function to get the exact omega for this data Phase 2: go back to the individual spectra and do a sequential fit with fixed omega determined in phase 1, and free phase and amplitude.

raquelalvarezbanos, 2016-02-03:
More from James:

> Equally someone should be looking into why MultiDomain fits scale badly (and perhaps also single domain fits with lots of parameters because of multiple peaks or other components), it looks like about N**3. Would a different choice of Minimiser help? Are Ties dealt with as efficiently as possible (e.g. if 10 spectra are fitted to a function with 5 parameters and 4 of these are set as global, is it a fit of 50, or 14, parameters internally?)

> At some point soon, someone will try to run CalMuonDetectorPhases on a CHRONUS file (606 detectors) and I'd guess it'll take about a day to run!

I think this should be done as a separate issue, so feel free to create a new one for this.

tom-perkins, 2016-02-03:
I've noticed a large difference between different data files. The fit runs quickly (20-30 seconds) when the spectra contain noticeable oscillations, but when the spectra do not have oscillations it takes many more iterations for the fit to converge.

For example `MUSR00022725` (used in the doc test), with oscillations, runs in 29 seconds on my machine, but `MUSR00022575`, which has no oscillations, took 7 minutes 7 seconds. (I fitted up to 4   microseconds for each).

raquelalvarezbanos, 2016-02-03:
That large difference between different data files makes sense, the runs James was using are EMU00020711.nxs and EMU00020721.nxs. He reported that running `CalMuonDetectorPhases` on the first one, after about 7 mins CPU time its progress bar was only at 4%.

PR #15191 (2016-02-16): Speed up CalMuonDetectorPhases
Resolves #15066 

**Changes:** 
Uses James's suggestion to speed up the `CalMuonDetectorPhases` algorithm:

1. Calculate asymmetry using an appropriate grouping and fit this to find the frequency
2. Use the resulting frequency as the fixed, shared value of `w` in a subsequent sequential fit of all spectra to `f(x) = A*sin(w*x + p)`

The sequential fit is done using `PlotPeakByLogValue` (as used in the MDF interface), rather than duplicating this functionality. However, this required adding the input workspace to the ADS and removing it at the end.

The detector g [... cut]
closed 2016-02-16 by raquelalvarezbanos