mantidproject/mantid #15103: Horizontal scrollbar in script window.

mantid-roman, 2016-01-26:
When there are long lines in the script window, it will not scroll to the end of them using the horizontal scroll bar.

NickDraper, 2016-04-04:
this is a known issue with qscintilla and also affects notepad2.
http://superuser.com/questions/326394/notepad2-horizontal-scrolling-issue

NickDraper, 2016-04-04:
There is a potential partial solution that I can try, as long as it does not affect performance too much.

NickDraper, 2016-04-04:
Note: The end of long lines can always be reached using the End key or the scrollbar arrows.

NickDraper, 2016-04-04:
The fix I tried was setting xxx on the scintilla component, but that does not work (it still acts that same as without the fix).

  //set QScintilla to track the max line length
  //the scroll width is adjusted to ensure that all of the lines currently 
  //displayed can be completely scrolled. 
  //This mode never adjusts the scroll width to be narrower.
  SendScintilla(SCI_SETSCROLLWIDTHTRACKING, 1);

I propose we leave this as a will not fix, with noting that pressing end or using the arrow keys is a work around.

closed 2016-05-10 by NickDraper