mantidproject/mantid #15153: Investigate scaling of multi-domain fits

tom-perkins, 2016-02-03:
Issue raised from discussion in issue #15066.

_Background:_
`CalMuonDetectorPhases` used a multi-domain function to fit a number of histograms in a dataset - the issue was this took a long time for large numbers of detectors (7 minutes for 96 histograms, longer for more...). Doing a sequential fit instead scales better with the number of detectors.

_Issue:_
This issue is to investigate how fitting with multi-domain functions scales with the number of histograms to be fitted. From James Lord:

> Equally someone should be looking into why MultiDomain fits scale badly (and perhaps also single domain fits with lots of parameters because of multiple peaks or other components), it looks like about N**3. Would a different choice of Minimiser help? Are Ties dealt with as efficiently as possible (e.g. if 10 spectra are fitted to a function with 5 parameters and 4 of these are set as global, is it a fit of 50, or 14, parameters internally?)

MichaelWedel, 2016-02-03:
How is this implemented for `CalMuonDetectorPhases`? I'm a bit curious, because the POLDI-fits use `MultiDomainFunction` with 400 spectra (500 points each) and ~40 parameters, the fit takes well below 20 seconds even for complex cases when a 5000 point spectrum is "wrapped" onto each spectrum.

tom-perkins, 2016-02-03:
Each spectrum is fitted to `f(x) = A*sin(w*x + p)`, where the parameter `w` is shared between all spectra. On investigating further this may have a dependence on the number of bins - fitting a shorter region of each spectrum is quite a bit faster. The number of spectra is usually around 96 but can be up to 606 on some instruments (CHRONUS).

MichaelWedel, 2016-02-03:
Okay, that doesn't sound so bad. You may be right with the dependency on the bins. For the fitting mode with 5000 points in each spectrum the major speedup was not calculating the function on the points where it's not needed. But for `sin` that will be difficult.

tom-perkins, 2016-04-04:
`CalMuonDetectorPhases` now uses a different strategy and has been sped up (see #15191), so this issue is no longer as urgent. There haven't been any other mentions of multi-domain fits being slow.
Removing milestone and setting as Low priority until the issue is raised again.

closed 2021-03-01 by stale[bot]