mantidproject/mantid #15155: 1D plots of MDHistoWorkspaces have values plotted at bin boundaries rather than centres

matthew-d-jones, 2016-02-03:
`MDHistoWorkspace.getLinePlot()` returns bin boundaries as x values, such that `num(x) == num(y)+1`. Qwt throws away the last `x` value and thus plots the `y` values against the lhs bin boundaries.

Plotting against bin centres makes more sense, such that `num(x) == num(y)`. This would also make omitting masked points (for #14781) trivial.

NB, `getLinePlot()` is used in ConvertMDHistoToMatrixWorkspace to convert 1D workspaces, which does require the bin boundaries.

Blocks #14781

matthew-d-jones, 2016-02-03:
The behaviour of `getLinePlot()` and `PlotMD` is strange for MDHistoWorkspaces and MDEventWorkspaces, see example script:
```python
# Create a 3D event workspace to test with
ws_mdew = CreateMDWorkspace(Dimensions='3', Extents='-2,2,-2,2,-2,2', Names='h,k,l',
                       Units='rlu,rlu,rlu', SplitInto='12')
FakeMDEventData(ws_mdew, UniformParams='10000', RandomizeSignal='1')
ws_mdew = SliceMD(ws_mdew,AlignedDim0='h,-2,2,12',AlignedDim1='k,-2,2,12',AlignedDim2='l,-2,2,12')

# Create projection
proj_id = Projection([1,1,0], [-1,1,0], [0,0,1])
proj_id.setType(0,'r')
proj_id.setType(1,'r')
proj_id.setType(2,'r')
ws_id = proj_id.createWorkspace()

# Cut once with NoPix true and once with NoPix false
out_md_1d_NoPix = CutMD(ws_mdew, Projection=ws_id, PBins=([-2,0.05,2],[-1.1,-0.9],[-0.1,0.1]), NoPix=True)
out_md_1d_Pix = CutMD(ws_mdew, Projection=ws_id, PBins=([-2,0.05,2],[-1.1,-0.9],[-0.1,0.1]), NoPix=False)

# Plot to compare (expect to look the same!)
plotMD(out_md_1d_NoPix, normalization=2, error_bars=True)
plotMD(out_md_1d_Pix, normalization=2, error_bars=True)
```

For MDHWs the value of the bin is plotted at the position of the lower bin boundary. For MDEWs 200 equally spaced points along the line are generated and the value of the bin that they fall in is plotted.

`getLinePlot()` can be called for an arbitrary line through a multidimensional workspace, however it is only currently used by PlotMD which only accepts workspaces with a single non-integrated dimension. The most solution would therefore be to restrict `getLinePlot()` to be dimension aligned through bin centres and plot the value of the bin at the position of the bin centre.

matthew-d-jones, 2016-02-03:
@AndreiSavici This was originally implemented by Janik Zikovsky, do you know if there is a design reason for `getLinePlot` to function for arbitrary lines, or have any other opinion on what the 1D plots should look like?

AndreiSavici, 2016-02-03:
There are many reasons why one might want to have non axis-aligned plots. One can get the original workspace in H,K,L, but want to be able to do a plot along HH direction. Or if one has Qx,Qy,Qz, and see that a peak is split (not always along the axes), they might want to see the distance between the peaks. However, all the cases I have seen mix only Q dimensions, not Q with E or something else.

matthew-d-jones, 2016-02-04:
@AndreiSavici Thanks.
In that case, if you have no objections, I will make `MDHW::getLinePlot` consistent with all other workspace types and return linearly spaced points.
The current implementation is used by `ConvertMDHistoToMatrixWorkspace` so I'll rename that to something like `getLineData`. If we want plots of bin-centre points in the future then that can be implemented using `getLineData`, or at least much of its code with a bit of refactoring.

AndreiSavici, 2016-02-04:
Sounds OK. The line plot makes sense only when rebinned anyway

PR #15187 (2016-02-08): Improvements to 1D plots of MDWorkspaces
Closes #15155 #15082 #14781

`MDHW::getLineData`, which is used by `plotMD`, is now consistent with other workspace types and returns linearly spaced points along the line. The previous implementation, which returns points at bin boundaries, is now `getLineData` and is used by the `ConvertMDHistoToMatrixWorkspace` algorithm.

Masked data was previously returned as NaN and appeared as a high value on the plot, now it is instead omitted from the plot. If masked data falls at one end of the plot then the range of the plot axis is reduced to the range of no [... cut]
closed 2016-02-08 by picatess