mantidproject/mantid #15188: Pixel segregation issue in ConvertToReflectometryQ

raquelalvarezbanos, 2016-02-05:
Jos sent the following script:

```
dref=s.OurLoad(r[36688])
dtrans=s.OurLoad(r[36689])

Trans=CreateTransmissionWorkspaceAuto(dtrans,AnalysisMode="MultiDetectorAnalysis",ProcessingInstructions="380-428",WavelengthMin=2.0)

IvsQ2, IvsLam2, thetaOut=ReflectometryReductionOneAuto(dref,AnalysisMode="MultiDetectorAnalysis",ProcessingInstructions="390-450", FirstTransmissionRun=Trans, ThetaIn="0.6",WavelengthMin=2.0,CorrectionAlgorithm='None',StrictSpectrumChecking=False)#,NormalizeByIntegratedMonitors=False
IvsQ, IvsLam, thetaOut=ReflectometryReductionOneAuto(dref,AnalysisMode="MultiDetectorAnalysis",ProcessingInstructions=','.join(map(str, range(250, 557))), FirstTransmissionRun=Trans, ThetaIn="0.6",WavelengthMin=2.0,CorrectionAlgorithm='None',StrictSpectrumChecking=False)#,NormalizeByIntegratedMonitors=False

TvL = ConvertSpectrumAxis(IvsLam,'SignedTheta')

ConvertToReflectometryQ(TvL,IncidentTheta=0.6 ,DumpVertexes=True,OverrideIncidentTheta=True, OutputWorkspace="Qmap",OutputVertexes="Qtable",Extents=[-5e-4,2e-4,0.008,0.1],NumberBinsQz=500,NumberBinsQx=500,Method='NormalisedPolygon',OutputAsMDWorkspace=False)
```

which produces a `QxQz` map with lines of dark intensity (we should obtain a smooth color map):

[image]

raquelalvarezbanos, 2016-02-08:
I've modified the script above to work with a smaller number of bins:

```
IvsQ, IvsLam, thetaOut=ReflectometryReductionOneAuto(dref,AnalysisMode="MultiDetectorAnalysis",ProcessingInstructions=','.join(map(str, range(400, 404))), FirstTransmissionRun=Trans, ThetaIn="0.6",WavelengthMin=2.0,WavelengthMax=4,WavelengthStep=0.5,CorrectionAlgorithm='None',StrictSpectrumChecking=False)#,NormalizeByIntegratedMonitors=False

TvL = ConvertSpectrumAxis(IvsLam,'SignedTheta')
ConvertToReflectometryQ(TvL,IncidentTheta=0.6 ,DumpVertexes=True,OverrideIncidentTheta=True, OutputWorkspace="Qmap",OutputVertexes="Qtable",Extents=[-8.9e-6,-7e-6,0.059,0.06],NumberBinsQz=1,NumberBinsQx=10,Method='NormalisedPolygon',OutputAsMDWorkspace=False)
```

This produces a `Qmap` with a single spectrum and 10 datapoints, where a point of low intensity is still seen:

[image]

I've looked at two of the quadrilaterals in more detail (see below) and it seems that for some reason the overlap between them is 1.9e-10, when it should be 0 (these two quadrilaterals are used to calculate the first datapoint in the figure above as `~ Y * overlap(inputQ,outputQ) / inputQ.area()`):

[image]

If I look at additional datapoints, the overlap between `inputQ` (black quadrilateral) and different `outputQ` should be 0 (see below) for the first three quadrilaterals, but it is 1.9e-10. For `outputQ6`, `outputQ7`, etc, the overlap should be `outputQ6`, `outputQ7`, etc, but it is 1.9e-10.

[image]

In summary, according to our `intersection` method:

`overlap(inputQ, outputQ1) = 1.9e-10`
`overlap(inputQ, outputQ2) = 1.9e-10`
`overlap(inputQ, outputQ3) = 1.9e-10`
`overlap(inputQ, outputQ4) = 1.99116e-11`
`overlap(inputQ, outputQ5) = 1.75475e-10`
`overlap(inputQ, outputQ6) = 1.9e-10`
`overlap(inputQ, outputQ7) = 1.9e-10`
`overlap(inputQ, outputQ8) = 1.9e-10`
`overlap(inputQ, outputQ9) = 1.9e-10`
`overlap(inputQ, outputQ10) = 1.9e-10`

raquelalvarezbanos, 2016-02-08:
I've modified the relevant bit of code to use boost::geometry and the overlaps I get are:

`overlap(inputQ, outputQ1) = 0`
`overlap(inputQ, outputQ2) = 0`
`overlap(inputQ, outputQ3) = 0`
`overlap(inputQ, outputQ4) = 1.97591e-11`
`overlap(inputQ, outputQ5) = 1.75405e-10`
`overlap(inputQ, outputQ6) = 1.9e-10`
`overlap(inputQ, outputQ7) = 1.9e-10`
`overlap(inputQ, outputQ8) = 1.9e-10`
`overlap(inputQ, outputQ9) = 1.9e-10`
`overlap(inputQ, outputQ10) = 1.9e-10`

raquelalvarezbanos, 2016-02-08:
I think the problem is in [Quadrilateral.cpp] (https://github.com/mantidproject/mantid/blob/35015ba9bac7dbc4b93c4de9f9068805d5e5968e/Framework/Geometry/src/Math/Quadrilateral.cpp#L96-L97):

```
bool Quadrilateral::contains(const ConvexPolygon &poly) const {
  return (minX() <= poly.minX() && poly.maxX() <= maxX() &&
          minY() <= poly.minY() && poly.maxY() <= maxY());
}
```

The above conditions are not sufficient to conclude that a polygon contains another polygon, I believe.

raquelalvarezbanos, 2016-02-12:
I spent the last few days working on this and identified potential causes of the problem (like the one I mention in my previous comment, some approximations we use in the code, etc), and I've come to the conclusion that none of them have an important impact on the results. However, today I noticed [this](https://github.com/mantidproject/mantid/blob/master/Framework/DataObjects/src/FractionalRebinning.cpp#L218-L223):

```
        if (removeOverlap) {
          const double overlapWidth =
              intersectOverlap.maxX() - intersectOverlap.minX();
          yValue *= overlapWidth;
          eValue *= overlapWidth;
        }
```

This block of code (which I do not understand) seems to have an important effect on the `QxQz` map. With this block of code, I get:

[image]

Without the `if` statement:

[image]

@OwenArnold what do you think about this?

OwenArnold, 2016-02-12:
@raquelalvarezbanos Good question! I haven't seen that before. This comes from FractionalRebinning, so probably @martyngigg has a better idea of why/when this was put in place first time around.

raquelalvarezbanos, 2016-02-15:
@martyngigg What we've just discussed gives the expected results (no dark "lines"), thanks!

martyngigg, 2016-02-15:
@raquelalvarezbanos That's good, glad it works now.

PR #15321 (2016-02-17): Pixel segregation issue
Fixes #15188 

Scientists reported a problem with `QxQz` map. They are getting dark regions that should not be there.

I identified two potential causes:
1. `Quadrilateral::contains` methods were not behaving as I would expect.
2. `FractionalRebinning::rebinToFractionalOutput` was not normalizing the `Y` values properly.

For tester:

1. Please review changes to `Quadrilateral::contains` (and new unit tests) and make sure you agree that these methods don't work for generic quadrilaterals (they work if our quadrilaterals are rectangles)

2. Run the script in #15188, and  [... cut]
closed 2016-02-17 by peterfpeterson