mantidproject/mantid #15199: Thread exhaustion in SliceViewer

OwenArnold, 2016-02-05:
While testing #15161 on OSX (although the platform shouldn't matter) I've found that I can invoke a "no thread available exception" see [description](http://www.appinf.com/docs/poco/Poco.NoThreadAvailableException.html) which propagates up through MantidPlot and results in termination options.

**To Reproduce**

* Create a large MDEventWorkspace
* View it in the SliceViewer
* Turn on Rebinning 
* Change the Rebinning parameters to yield a largeish MDHistoWorkspace. 
* Switch it to Auto Rebin
* Now use sliders and zoom in a lot. You should be racking-up calls to BinMD on lots of different threads.
* Keep doing this and you will eventually get the aforementioned exception.

I see this as protecting against Machiavelli rather than Murphy as the user would have to deliberately and  for a sustained period overstress the tool to get this issue. The fix involves making changes to AlgorithmRunner i'm uncomfortable with this late in the release cycle. That's why i've made it a 3.7 issue despite the fact that it's a bugfix.

**How to Fix**

We can probably fix this by addressing the `startAlgorithm` method [here](https://github.com/mantidproject/mantid/blob/master/MantidQt/API/src/AlgorithmRunner.cpp#L72-L89). I suggest making this return a boolean to indicate whether the AlgorithmRunner was actually able to start  the algorithm. I'd wrap the call to the async execution in a try catch and try to swallow any `NoThreadAvailableExceptions`. If an exception of this type is thrown we should return false. We need Anton's fix #15161 in place to ensure that a failed BinMD invocation does not result in an attempt to draw the overlay.

AntonPiccardoSelg, 2017-10-05:
@Mantid-Matthew This might be another good issue to get working on the SliceViewer. Best to talk to @martyngigg about the underlying cause.

Mantid-Matthew, 2019-03-21:
Slice viewer is being rewritten for the workbench and this is an issue users are unlikely to see.

closed 2019-03-21 by Mantid-Matthew