mantidproject/mantid #15206: Remove UserAlgorithms + associated content from packages

martyngigg, 2016-02-08:
It was decided at the 2016 meeting that the usage of user algorithms is no longer high enough to justify shipping them with the packages. The directory will now become a set of examples and will not be shipped with the package at this time.

The associated content such as `boost` headers etc can also be removed from the packages.

martyngigg, 2016-02-10:
@ElliotAOram I'd like you to look at this as part of maintenance now that you can build the windows package. It's probably best to have a chat about it when you're ready to start it

ElliotAOram, 2016-02-17:
Build statistics with branch from 17/02/2016:

|  Build Time(`mm:ss`)  | Install Time(`mm:ss`)  |  Size(`MB`): |
|:---------------------------:|:----------------------------:|:---------------:|
|             ~10:35            |                00:51            |        911       |

ElliotAOram, 2016-02-17:
Statistics after changes have been made:

|  Build Time (`mm:ss`)  | Install Time (`mm:ss`)  |  Size (`MB`)  |   Size on Disk (`MB`)  |
|:----------------------------:|:-----------------------------:|:----------------:|:---------------------------:|
|             ~07:40             |                00:37             |        789        |               821               |
|             ~07:04             |                00:37             |        789        |               821               |
|                                     |                00:37             |        789        |               821               |

martyngigg, 2016-02-17:
@ElliotAOram Is that using a debug or release build? The size is quite a lot larger than I would expect.

ElliotAOram, 2016-02-17:
@martyngigg I thought it was release but I can double check. Just to clarify this is the size of the MantidInstall directory after installation.

martyngigg, 2016-02-17:
Ah I misunderstood, sorry. What is the size of the package itself?

ElliotAOram, 2016-02-17:
Old = 200MB
New = 183MB

martyngigg, 2016-02-17:
I'm still surprised at the build time. Can you see any obvious way of telling what takes the time? I think `cpack` as `--debug` & `--verbose` flags.

ElliotAOram, 2016-02-17:
Running with `--verbose` shows a long wait time on the following:
```
CPack: Create package
CPack Verbose: Package files to: C:/Users/HVQ65228/Documents/Build/Secondary/_CPack_Packages/win64/NSIS/mantid-3.6.20160217.1059-win64.exe
CPack Verbose: Configure file: C:/Users/HVQ65228/Documents/GitHub/secondary_mantid/buildconfig/CMake/NSIS.template.in to C:/Users/HVQ65228/Documents/B
uild/Secondary/_CPack_Packages/win64/NSIS/project.nsi
CPack Verbose: Execute: "C:/Program Files (x86)/NSIS/makensis.exe" "C:/Users/HVQ65228/Documents/Build/Secondary/_CPack_Packages/win64/NSIS/project.nsi"
```

With the `--debug` option this same wait time happens after:
```
C:/cygwin/home/dashboard/CMakeReleaseDirectory/650c630a/Source/CPack/cmCPackGenerator.cxx:1012 cmCPackNSISGenerator::SetOption(CPACK_NSIS_INSTALLER_MU
I_ICON_CODE, !define MUI_ICON "C:/Users/HVQ65228/Documents/GitHub/secondary_mantid/images\MantidPlot_Icon_32offset.ico"
!define MUI_UNICON "C:/Users/HVQ65228/Documents/GitHub/secondary_mantid/images\MantidPlot_Icon_32offset.ico"
)
C:/cygwin/home/dashboard/CMakeReleaseDirectory/650c630a/Source/CPack/cmCPackGenerator.cxx:1012 cmCPackNSISGenerator::SetOption(CPACK_NSIS_INSTALLER_IC
ON_CODE, !define MUI_HEADERIMAGE_BITMAP "C:/Users/HVQ65228/Documents/GitHub/secondary_mantid/images\MantidPlot_Icon_32offset.png"
)
C:/cygwin/home/dashboard/CMakeReleaseDirectory/650c630a/Source/CPack/cmCPackGenerator.cxx:1012 cmCPackNSISGenerator::SetOption(CPACK_NSIS_INSTALLATION
_TYPES, )
C:/cygwin/home/dashboard/CMakeReleaseDirectory/650c630a/Source/CPack/cmCPackGenerator.cxx:1012 cmCPackNSISGenerator::SetOption(CPACK_NSIS_INSTALLER_MU
I_COMPONENTS_DESC, )
C:/cygwin/home/dashboard/CMakeReleaseDirectory/650c630a/Source/CPack/cmCPackGenerator.cxx:1012 cmCPackNSISGenerator::SetOption(CPACK_NSIS_PAGE_COMPONE
NTS, )
C:/cygwin/home/dashboard/CMakeReleaseDirectory/650c630a/Source/CPack/cmCPackGenerator.cxx:1012 cmCPackNSISGenerator::SetOption(CPACK_NSIS_FULL_INSTALL
, File /r "${INST_DIR}\*.*")
C:/cygwin/home/dashboard/CMakeReleaseDirectory/650c630a/Source/CPack/cmCPackGenerator.cxx:1012 cmCPackNSISGenerator::SetOption(CPACK_NSIS_COMPONENT_SE
CTIONS, )
C:/cygwin/home/dashboard/CMakeReleaseDirectory/650c630a/Source/CPack/cmCPackGenerator.cxx:1012 cmCPackNSISGenerator::SetOption(CPACK_NSIS_COMPONENT_SE
CTION_LIST, )
C:/cygwin/home/dashboard/CMakeReleaseDirectory/650c630a/Source/CPack/cmCPackGenerator.cxx:1012 cmCPackNSISGenerator::SetOption(CPACK_NSIS_SECTION_SELE
CTED_VARS, )
```

The `makensis.exe` process is still running at ~12-13% CPU usage at this time

ElliotAOram, 2016-02-18:
Test build with different NSIS compressors:

| Compression  |  Build Time  | Install Time  |  Size  |   Size on Disk   |  Installer Size  |
|:------------------:|:---------------:|:---------------:|:-------:|:-------------------:|:------------------:|
|  default(ZLIB)  |     07:40       |       00:37     |  789   |         821          |        183          |
|  default(ZLIB)  |     07:04       |       00:37     |  789   |         821          |        183          |
|        BZIP2       |     03:00       |       00:42     |  789   |         821          |        229          |
|        BZIP2       |     02:59       |       00:43     |  789   |         821          |        229          |
|        LZMA       |     07:00       |      00:37      |  789   |         821          |        183          | 
|        LZMA       |                     |                     |           |                         |                        |

The results suggest that the default compression is actually LZMA not ZLIB as was suggested by the documentation 
While attempting to use the ZLIB compressor I get the the following:
```
CPack Verbose: Execute: "C:/Program Files (x86)/NSIS/makensis.exe" "C:/Users/HVQ65228/Documents/Build/Secondary/_CPack_Packages/win64/NSIS/project.nsi
"
CPack Error: Problem running NSIS command: "C:/Program Files (x86)/NSIS/makensis.exe" "C:/Users/HVQ65228/Documents/Build/Secondary/_CPack_Packages/win
64/NSIS/project.nsi"
Please check C:/Users/HVQ65228/Documents/Build/Secondary/_CPack_Packages/win64/NSIS/NSISOutput.log for errors
CPack Error: Problem compressing the directory
CPack Error: Error when generating package: mantid
```

ElliotAOram, 2016-02-18:
Something further to potentially investigate is that the dictionary size for LZMA can be changed at the moment it is set to 8MB as a default. I believe this could explain why the compression is so slow by comparison to the others.

martyngigg, 2016-02-18:
Thanks, @ElliotAOram. Those results are really interesting. Could you create a separate issue for us to continue with this exploration. If we can get the build time to 3 minutes and the install time to ~20s then that really is worth it.

PR #15373 (2016-02-22): Remove UserAlgorithms from packages
Fixes #15206

**This can only be tested on windows**

The `UserAlgorithms` package should no longer be built. The  user algorithms still exist, they can now be found in `Framework/Examples`.

# To Test
* Ensure that all unit test run and pass 
* Ensure that the User algorithms directory (`Framework/UserAlgorithms`) is no longer present in the source.
* Ensure the `Examples` directory is in the Framework directory of the source code and it contains 6 `.cpp` algorithms (plus their respective headers)
* Run CMake ensuring you check the option `ENABLE_CPACK`
* I [... cut]
closed 2016-02-22 by peterfpeterson