mantidproject/mantid #15237: Keyboard control of SpectrumViewer

peterfpeterson, 2016-02-09:
Keyboard control does not appear to work anymore. To reproduce try any of the following:
* Change `X Min`, `X Max`, or `Step(-if Log)` and get it to update the view
* Click on the main image then move the cursor to change where it is pointing. The horizontal and vertical graphs should update.

AndreiSavici, 2016-02-09:
Just some quick comments:
* once you click on the main image, move cursor using arrow keys
* the cursor moves, but the plots on the x and y axis do not  update
* I found the same behaviour in the previous release
* in MantidPlot 3.4.1, I cannot move the cursor  at all in the vertical direction, but it works as expected (including spectrum update) when I move left and right

martyngigg, 2016-02-10:
Confirmed both bullets on Ubuntu 14.04 too.

mantid-roman, 2016-02-10:
If you click the image then press Enter the plots will update using the keyboard. Press Enter again (or Esc) updating stops. Maybe it's a good thing.

martyngigg, 2016-02-10:
@mantid-roman That works for me too. It feels like something more visual like a checkbox for auto-updating would provide a better experience.

Does changing the X range do anything for you?

mantid-roman, 2016-02-10:
Agree.

Changing x range doesn't change the image, only after changing intensity. This should be fixed.

peterfpeterson, 2016-02-10:
The (apparently long broken) functionality is that changing the binning and pressing enter updated the image and related plots, moving the cursor (using keyboard) updated the plots and the tables.

peterfpeterson, 2016-02-10:
@mantid-roman You need to branch from `release-3.6.0` if you don't want to make it hard to merge.

mantid-roman, 2016-02-10:
I forgot

peterfpeterson, 2016-02-10:
That branch can be turned into a PR for merging into `master`.

peterfpeterson, 2016-02-10:
Your commit fixed the Xmin/Xmax issue. Updating spectra is still broken.

peterfpeterson, 2016-02-10:
For the arrow key issue I think 5ac1813624689ea082064b884bad6e1ae6e103dd matters most.

mantid-roman, 2016-02-11:
Now we have 3 branches :)

peterfpeterson, 2016-02-11:
The tip of the branch is really close.

peterfpeterson, 2016-02-11:
Only two branches. @mantid-roman made both.

mantid-roman, 2016-02-11:
Are we ready for PR?

PR #15272 (2016-02-11): Spectrum viewer keyboard control
Fixes #15237.

To test check that the functionality of the Spectrum Viewer is how it is supposed to be.
closed 2016-02-11 by peterfpeterson