mantidproject/mantid #15268: Add more forward declarations for typedefs in Kernel and Geometry

FedeMPouzols, 2016-02-11:
After #15246 is done, consider adding forward declarations (in `_fwd.h` header files) for some commonly used typedefs which are currently inside lengthy headers. 

Good candidates ranked by benefit (first rough guess):
- Kernel::Unit
- Geometry::IComponent
- Geometry::Instrument
- Geometry::IDetector

And then replace the includes pointing to the lengthy headers with the new `_fwd.h` headers.

As a possible bonus: take `MantidVec` out of `MantidKernel/cow_ptr.h`.

These types are used in an apparently significant percentage of algorithms and data-handlers.
Initially I'd estimate a not so great benefit/effort ratio from these changes (compared to #15246), so I don't think this is a high priority for 3.7 maintenance.

FedeMPouzols, 2016-02-16:
Forward declarations have been introduced for Instrument and IDetector in #15320. 
In this issue these should be used more widely. A few example good candidates to use forward declarations of Instrument:
```
Framework/DataObjects/inc/MantidDataObjects/Peak.h
Framework/DataObjects/inc/MantidDataObjects/SpecialWorkspace2D.h
Framework/DataObjects/inc/MantidDataObjects/PeaksWorkspace.h
```
But what remains to be done with regards to this does not seem to have a big potential benefit.
For `Unit`, some candidate places to use forward declarations (when needed, some of them seem not to use it at all):
```
Framework/API/inc/MantidAPI/IFunctionMW.h
Framework/API/inc/MantidAPI/ParamFunction.h
Framework/API/inc/MantidAPI/IFunction.h
Framework/API/inc/MantidAPI/IFunctionMD.h
Framework/API/inc/MantidAPI/TextAxis.h
Framework/API/inc/MantidAPI/Axis.h
Framework/API/inc/MantidAPI/SpectraAxis.h
```
It could probably stay in Axis.h but see if it can go away from the other headers.

NickDraper, 2019-07-25:
Closing for now as benefit is not large enough

closed 2019-07-25 by NickDraper