mantidproject/mantid #15269: Zooming in Instrument View stops working after drawing shape

mantid-roman, 2016-02-11:
* Draw a shape in Draw tab.
* Change to Render
* Switch projection (render mode) eg from the 3D view to a flat one
* Zooming isn't working
* It comes back after changing tabs

OwenArnold, 2016-02-11:
According to #14914 this is a regression, so I've raised the priority. As we have users working from the nightly-builds as the SNS, we need to fix this asap.

LamarMoore, 2016-02-12:
I have identified this issue as present in version 3.5. I am checking now to see if it is also present in 3.6. I do not think it is related to the refactoring of the instrument view code. I will confirm shortly.

LamarMoore, 2016-02-12:
I can confirm that this issue was not created in the process of refactoring the instrument view code. This is present in both 3.5 and 3.6.

jclarkeSTFC, 2024-07-18:
Still a problem in 6.10