mantidproject/mantid #15279: FFT: deal with phase when input signals do not start at X=0

tom-perkins, 2016-02-12:
This issue is the second stage of #15264 

Muon scientists would like `FFT` to handle input signals which do not start at X=0. The idea is to be able to add a phase to the data. Ask James Lord for further information.

James says:
> I've just noticed that an ordinary FFT will happily accept a workspace with uniform sized bins (tidied by Rebin) that doesn't start at x=0 or even have an exact x=0 point anywhere in the spectrum or extrapolated from either end. That's OK as it is, but the output phases seem not to depend on the x starting point. For example:

> ``` python
Rebin(InputWorkspace='PhaseQuadOutput', OutputWorkspace='Binned', Params='0.016')
FFT(InputWorkspace='Binned', OutputWorkspace='FFT1', Imaginary=1) 
ScaleX(InputWorkspace='Binned', OutputWorkspace='Offsetted', Factor='1', Operation='Add') # 1us offset is 62.5 bins
FFT(InputWorkspace='Offsetted', OutputWorkspace='FFT2', Imaginary=1)
```

>and then FFT1 and FFT2 compare equal, when they shouldn't! (there should be a phase difference in spectra 0 and 1, however spectrum 2=magnitude will still be equal)

>So any analysis looking for a peak in the FFT output and then setting up a sinewave to fit or subtract from the time dependent data is going to get its phase wrong. Maxent may suffer similarly. The test will be to Crop the data slightly differently, perhaps keeping the same number of points so the resolution is unchanged, and the phases of peaks should not vary by more than random statistical errors.

raquelalvarezbanos, 2016-02-12:
I think the input property "Shift" could be what they need, an example script:

```
from math import pi, sin, cos
X = []
YRe = []
YIm = []
E = []
N = 1000
w = 10
for i in range(0,N):
       x = 2*pi*i/N
       X.append(x)
       YRe.append(cos(w*x))
       YIm.append(sin(w*x))
       E.append(0.1)
ws = CreateWorkspace(DataX=X+X,DataY=YRe+YIm,DataE=E+E,NSpec=2)

FFT(InputWorkspace='ws', OutputWorkspace='FFT1', Imaginary=1)

factor = pi/w
ScaleX(InputWorkspace='ws', OutputWorkspace='Offsetted', Factor=factor, Operation='Add')
FFT(InputWorkspace='Offsetted', OutputWorkspace='FFT2', Imaginary=1, Shift=factor)
```

tom-perkins, 2016-03-08:
`FFT` should calculate and apply the shift automatically. This could be done by a true/false property called "Autoshift" or similar - the existing "Shift" should be left as a number to not break existing scripts.

closed 2016-03-31 by FedeMPouzols