mantidproject/mantid #15281: Resolve issues for CPPcheck 1.72

NickDraper, 2016-02-12:
http://builds.mantidproject.org/view/Static%20Analysis/job/cppcheck-1.72/

raquelalvarezbanos, 2016-02-16:
I only spotted an unused member variable (https://github.com/mantidproject/mantid/pull/15340). Many of the issues that were picked up by cppcheck1.71 are not real issues (see for instance [this](http://builds.mantidproject.org/view/Static%20Analysis/job/cppcheck-1.72/1019/cppcheckResult/source.34/#480) or [this](http://builds.mantidproject.org/view/Static%20Analysis/job/cppcheck-1.72/1019/cppcheckResult/source.26/#56)). The remaining ones, I am not really sure, so I am leaving this ticket unassigned in case other developers want to work on this.

quantumsteve, 2016-02-18:
@AndreiSavici How did you recommend changing the V3D and VMD wrappers that both provides better documentation from python and eliminates the cppcheck warnings?

AndreiSavici, 2016-02-19:
See https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/mantid/kernel/src/Exports/Quat.cpp
```
      .def("__add__", &Quat::operator+, (arg("left"), arg("right")))
      .def("__iadd__", &Quat::operator+=, boost::python::return_self<>(),
           (arg("self"), arg("other")))
```

PR #15438 (2016-03-04): cppcheck 1.72 errors 
Work to clear the errors for the new version of cppcheck 1.72

**To test:**
1. look at the output of the cppcheck 1.72 job -> currently pointed at this branch http://builds.mantidproject.org/view/Static%20Analysis/job/cppcheck-1.72/
2. The existing cppcheck job will report errors (there is incompatibility between the two versions)

Fixes #15281 

 "Does not need to be in the release notes."

## On release/Merge
1. The existing CPP check jobs will need to be updated to 1.72
2. The cppcheck 1.72 job should be deleted (it currently points to this branch)

---

#### Reviewer  [... cut]
closed 2016-03-04 by FedeMPouzols