mantidproject/mantid #15288: C++11 use std::mutex

NickDraper, 2016-02-12:
Move Poco::Mutex, Poco::FastMutex, boost::mutex,... to std::mutex.

picatess, 2016-02-17:
Scoping out the work:

__`Poco::Mutex` --> `std::recursive_mutex`__

[`Poco::Mutex`] (http://www.appinf.com/docs/poco/Poco.Mutex.html) appears to be recursive and hence maps into [`std::recursive_mutex`](http://en.cppreference.com/w/cpp/thread/recursive_mutex)

__`Poco::FastMutex` --> `std::mutex`__

[`Poco::FastMutex`](http://www.appinf.com/docs/poco/Poco.FastMutex.html) is not recursive and hence maps into [std::mutex](http://en.cppreference.com/w/cpp/thread/mutex)

__`Poco::ScopedLock` --> `std::lock_guard`__

[`Poco::ScopedLock`](http://www.appinf.com/docs/poco/Poco.ScopedLock.html) is a load-and-forget RAII-style lock which should map to [std::lock_guard](http://en.cppreference.com/w/cpp/thread/lock_guard)

NickDraper, 2016-02-18:
Anton,  It is possible that Poco::Mutex was used in places when FastMutex would have sufficed.

PR #15377 (2016-02-25): Repleace Poco::Mutex, Poco::FastMutex with elements from header <mutex>
Fixes #15288 

__Context__

The aim of this issue is to move `Poco` mutexes to `std` mutexes. The way the two systems map is described in the issue itself.  Note that anything associated with `Poco/RWLock` was not touched.

### For tester

* Code review carefully. I did leave the original structure in place. In some cases I changed the manual locking to either use a `lock_guard` or a `unique_lock`. Assure yourself that it makes sense in these situations. Note that we should in the future probably replace the manual, direct [... cut]
closed 2016-02-25 by FedeMPouzols