mantidproject/mantid #15293: Clang working on linux - singleton stopping initializing python

NickDraper, 2016-02-12:
Talk to @quantumsteve Steven Hahn for details

rosswhitfield, 2019-03-05:
It works, build can be found at http://builds.mantidproject.org/job/master_clean-ubuntu-18.04-clang/

closed 2019-03-05 by rosswhitfield

quantumsteve, 2019-03-05:
Is this another singleton issue?
```
20:01:53 2262/2299 Test #1912: mantidqt_qt5.test_multifileinterpreter.test_multifileinterpreter .............................................................***Failed    1.86 sec
20:01:53 FrameworkManager-[Notice] Welcome to Mantid 3.13.20190304.2225
20:01:53 FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
20:01:53 Traceback (most recent call last):
20:01:53   File "/home/builder/Jenkins/workspace/master_clean-ubuntu-18.04-clang/Framework/PythonInterface/test/testhelpers/testrunner.py", line 157, in <module>
20:01:53     main(sys.argv)
20:01:53   File "/home/builder/Jenkins/workspace/master_clean-ubuntu-18.04-clang/Framework/PythonInterface/test/testhelpers/testrunner.py", line 113, in main
20:01:53     test_module = imp.load_source(module_name(pathname), pathname)
20:01:53   File "/home/builder/Jenkins/workspace/master_clean-ubuntu-18.04-clang/qt/python/mantidqt/widgets/codeeditor/test/test_multifileinterpreter.py", line 21, in <module>
20:01:53     from mantidqt.widgets.codeeditor.multifileinterpreter import MultiPythonFileInterpreter
20:01:53   File "/home/builder/Jenkins/workspace/master_clean-ubuntu-18.04-clang/qt/python/mantidqt/widgets/codeeditor/multifileinterpreter.py", line 21, in <module>
20:01:53     from mantidqt.widgets.codeeditor.scriptcompatibility import add_mantid_api_import, mantid_api_import_needed
20:01:53   File "/home/builder/Jenkins/workspace/master_clean-ubuntu-18.04-clang/qt/python/mantidqt/widgets/codeeditor/scriptcompatibility.py", line 19, in <module>
20:01:53     from mantid import simpleapi
20:01:53   File "/home/builder/Jenkins/workspace/master_clean-ubuntu-18.04-clang/build/bin/mantid/simpleapi.py", line 1423, in <module>
20:01:53     _api.FrameworkManagerImpl.Instance()
20:01:53 RuntimeError: Exists
```

rosswhitfield, 2019-03-05:
Looks like it, here is the previous fix that @martyngigg did https://github.com/mantidproject/mantid/pull/23188