mantidproject/mantid #15295: Fix bug in vtkMDHexFactory which causes 3D slices of higher dimensional workspaces to not contain any points

matthew-d-jones, 2016-02-12:
vtkMDHexFactory expects zero thickness slices to contain data, due to the strange implementation of MDPlanes before #15209. Previous to that PR it would actually return a slice 2 bins thick if the two MDPlanes defining the slice coincide with a bin boundary and 1 bin thick otherwise.

Defining the higher bound plane (the one with the -ve normal) as lying on the next highest bin boundary would solve the problem. NB efficiency is important as the slicing procedure is repeated many times, for example to create a movie when the 4th dimension is time.

martyngigg, 2016-02-12:
@matthew-d-jones Is this the cause of the performance test segfault?

matthew-d-jones, 2016-02-12:
@martyngigg yes

martyngigg, 2016-02-12:
Good that you found that. Is it possible to implement a standard unit test to show the bug rather than it showing up in the performances tests?

matthew-d-jones, 2016-02-12:
Yes, will do.

PR #15335 (2016-02-16): Fix time slicing for VSI
Fixes #15295 

Fix is to create time slices which are centred on bin centres and are one bin thick.

**To Test:**

.deb is at `smb://olympic/babylon5/Scratch/Matthew%20Jones/15335`

You can use the following script to make a 4D workspace.

```python
ws_mdew = CreateMDWorkspace(Dimensions='4', Extents='-2,2,-2,2,-2,2,-2,2', Names='h,k,l,t',
                                                 Units='rlu,rlu,rlu,s', SplitInto='15')
for peak_pos in np.linspace(-1.875, 1.875, 16):
    FakeMDEventData(ws_mdew, PeakParams='1000,0,0,'+str(peak_pos)+','+str(peak_pos)+',0.005')
    [... cut]
closed 2016-02-16 by picatess