mantidproject/mantid #15319: Explore more ways to reduce number of recursive includes in Algorithm.h (with desire of speeding up builds)

FedeMPouzols, 2016-02-15:
In #15246, this
```
Algorithm.h: 155 includes
MatrixWorkspace.h: 138 includes
ExperimentInfo.h: 91 includes
```
has been reduced to:
```
Algorithm.h: 79 includes
MatrixWorkspace.h: 119 includes
ExperimentInfo.h: 69 includes
```

Several points should be explored further, possibly in different issues/branches:

- `MatrixWorkspace` and maybe `ExperimentInfo` still look bloated. 

- Geometry related forward declares introduced in #15246 should be used more throughout the code. Maybe more forward declares should be included. Geometry headers tend to be heavy and they are unnecessarily (over)-included from may header and implementation files.

- `ISpectrum.h` could/should be removed from `MatrixWorkspace.h`?

- Includes In Algorithm.h:
  - WorkspaceProperty.h  is really heavy, with lots of inline code.
  - Properties and Progress seem to be reasonable things to include
  - AlgorithmFactory is required especially for the `DECLARE_ALGORITHM()` macro, but maybe not all of it should be included by every algorithm.

- Try to make sure that geometry stuff is not included unnecessarily.

SimonHeybrock, 2016-02-15:
@FedeMPouzols Great! Have a look at my changes in https://github.com/mantidproject/mantid/pull/15324. Basically this should allow us to remove any includes of `Instrument` or `Detector` (and the like) from (in this case) `ApplyTransmissionCorrection`.

FedeMPouzols, 2016-02-15:
@SimonHeybrock : I'm sure something like #15324 would improve a lot the situation if used all throughout. I have the feeling that only a tiny fraction of our code really needs to manipulate the fully feature geometry objects. For example, there's no few places in the code where Instrument is included just to retrieve the name of the instrument!

SimonHeybrock, 2016-02-15:
@FedeMPouzols Yes, you are right. I would like to try and see how much impact this has. Is there a reasonably easy way of stripping down the includes that go into `ApplyTransmissionCorrection`, such that we could compare compile times?

FedeMPouzols, 2016-02-15:
I'll have to check that one. Avoiding the include of MantidGeometry/Instrument.h, replaced by MantidGeometry/IDetector.h cut the compilation time from 5.07 down to 4.66.

martyngigg, 2019-10-22:
Good work been done. Closing.

closed 2019-10-22 by martyngigg