mantidproject/mantid #15337: Engg GUI advanced calibration: output names fixes and improvements

FedeMPouzols, 2016-02-16:
When you use cropped calibration giving a list of spectra, you need to be able to give the "custom" bank name (from #14591). This could go in a text box that is only enabled when you select a custom range of spectra (sibling of the current text box where you can enter the spectra numbers).

Possible issues. When you run cropped calibration, selecting either bank 1 or bank 2, the interface seems to be saving both .his files:
```
SaveOpenGenieAscii started
SaveOpenGenieAscii successful, Duration 0.04 seconds
Saved focused workspace as file: /home/fedemp/EnginX_Mantid/User/RB0000322/Calibration/ob+ENGINX_236516_North_bank.his
SaveOpenGenieAscii started
SaveOpenGenieAscii successful, Duration 0.04 seconds
Saved focused workspace as file: /home/fedemp/EnginX_Mantid/User/RB0000322/Calibration/ob+ENGINX_236516_South_bank.his
```
Check this and fix. Related: #14594.
I noticed this typo: "intgration" in a string::find in the presenter which might be related.

FedeMPouzols, 2016-02-16:
@shahrozahmed1 : we should check this tomorrow after the ENGIN-X update.

shahrozahmed1, 2016-02-23:
```
The calibration calculations failed. Some input properties were not valid. See log messages for details. 
The cablibration did not finish correctly.
```
 Sometimes `Cropped Calibration` while utilising `SpectrumIDs` fails which can also be looked at along with this issue.

closed 2016-03-05 by quantumsteve