mantidproject/mantid #15343: Indirect Data Analysis I(Q,t) Fit - Running the interface does not update miniplot

ElliotAOram, 2016-02-16:
In the `I(Q, t) Fit` interface the mini plot is not being correctly updated after the run algorithm is called. This works for Fit single Spectrum. This is most likely caused by the workspace not being found in the ADS due to incorrect name.

ElliotAOram, 2016-03-01:
The above are all not related to this issue and are incorrectly referenced

martyngigg, 2016-03-01:
Sorry, @ElliotAOram I meant to type 15434!

ElliotAOram, 2016-03-01:
@martyngigg not a problem!

ElliotAOram, 2016-03-01:
This issue is caused by using the new `iris` files rather than the old `irs` files. Currently the underlying code gets the instrument prefix short version to name the output hence using `iris` files still produces files beginning with `irs`. This issue will be solved in #15440 as the OutputWorkspace names will be set by the user rather than automatically generated.

closed 2016-05-16 by FedeMPouzols